aseba-community/aseba #358: Flicker when not in full screen

motib, 2015-02-11:
Opening a program when VPL is not in full screen causes severe Flicker. Going to full screen stops the flicker. Strangely, it happens only in one specific program.
Film clip: https://dl.dropboxusercontent.com/u/13792989/flicker.mp4
VPL program: https://dl.dropboxusercontent.com/u/13792989/test.aesl

stephanemagnenat, 2015-02-11:
Thank you for reporting this bug, I saw it already, and I am surprised it is not here.

There is a problem in the delay of resize events within Qt and the layouting of the view. I tried to make that work (see for instance commit b4bfd94a9633c3f1afa4ff6bc2c9cd34c1375af7).

Fixing this bug is in the TODO list of @sampla.

closed 2015-02-11 by stephanemagnenat

stephanemagnenat, 2015-02-11:
@motib please test with tomorrow's nightly build whether you are happy with the correction.

motib, 2015-02-12:
The latest build I see is 11 February 2015 01:17 d7b432b which I downloaded yesterday (and the time stamp is before your comment about "tomorrow"). Is there some reason that the build isn't run every night??

mbonani, 2015-02-12:
We made a built yestrrday morning it should be in the list. Perhaps the
arrangement do not take care of the hour of the built

Michael Bonani

Excuse my brevity sent from my smartphone

motib, 2015-02-12:
Strangely, the 09:49 build is shown _below_ the 01:17 build and I just looked at the top one, assuming it to be the latest one.

The flicker no longer occurs.

But, I don't understand "always show the scrollbar". For the program test.aesl it appears in the small window in the AESL pane but not in the VPL pane, while in fullscreen the opposite occurs. In both cases, the scrollbar appears only if it is really needed, so there is no bug and I'll close the issue.

Thanks

PR #359 (2015-02-11): always show scrollbar to prevent flicker
Fixes #358