aseba-community/aseba #360: Visual error indicator does not update properly

stephanemagnenat, 2015-02-11:
In some cases, the visual error indicator does not update properly. Here is how to reproduce:

* On line 1, drop a button event, the missing action indicator is updated
* On line 2, drop a color action
* On line 1, drop a color action
The information indicator changes to line 2, but the triangular indicator stays on line 1 until the next refresh.

If you now click on the "New" button, the information mark will stay in the scene until a next change.

motib, 2015-02-12:
Stephane: what do you mean by "the next refresh"? I see the triangle move to line 2 immediately.

I was able to reproduce the second part of the issue.

In my opinion, the red "!" marker is information overload. There is already: (1) the red error message, (2) the red border around the offending pair, (3) the red triangle. That should be enough :-).

stephanemagnenat, 2015-02-12:
I mean something that generates a refresh of the window, such as a resize event, a hide/show event, or another change in the program. It might well be that this problem does not appear on Windows, as the exact policy of propagating refresh events might be system-specific. Yet I think that a request to update is missing, causing these problems.

stephanemagnenat, 2015-02-12:
The goal of red "!" and the line is to visually mark the locations of errors.

sampla, 2015-03-02:
I am sorry, but I cannot reproduce the issue and have tried with many different window sizes.

stephanemagnenat, 2015-03-09:
Maybe it is OS X specific then.

stephanemagnenat, 2015-03-17:
It definitely seems to be specific to OS X, as I have seen other artefacts with that OS.

stephanemagnenat, 2015-03-17:
First part fixed in commit 2187364.

mbonani, 2015-03-17:
[image]

There is a similar bug for the red circle for similar events (i put 3 the same and correct one)
also on windows. It depents aslo of the refresh state. Tested with last commit 2187364

stephanemagnenat, 2015-03-17:
Very interesting. What steps did you do, the ones I put in the initial bug report, or other ones?

stephanemagnenat, 2015-03-17:
I believe that I found the cause of the bug. The `boundingRect()` for `EventActionsSet` does not extend to the positions of the error markers. Qt has no obligation to redraw beyond the `boundingRect()` of `QGraphicsItem`, see http://doc.qt.io/qt-4.8/qgraphicsitem.html#boundingRect

stephanemagnenat, 2015-03-17:
The original issue is now fixed in commit 086c31f. @mbonani could you check if it corrects your problem as well?

mbonani, 2015-03-17:
still there it is a different one.

stephanemagnenat, 2015-03-17:
Ok, could you give me the exact the to reproduce the bug?

mbonani, 2015-03-17:
Put 3 indentical events with 3 actions valid
The third one has to be not visible you have to scroll to see it
Then change the middle event so that the third one is the falling one
In some case depends on your scroll the red circle is present still a
refresh appens

Michael Bonani

Excuse my brevity sent from my smartphone

stephanemagnenat, 2015-03-17:
I believe you but on OS X 10.10.2 I could not reproduce the bug: following your procedure leads to a perfectly correct behaviour.

motib, 2015-03-18:
OK In nightly-2015-03-18-aseba-bin-1.3.9-git-842463b-win32.exe

stephanemagnenat, 2015-03-18:
@mbonani could you have other people at Mobots/Mobsya try to reproduce the problem you have?

mbonani, 2015-03-18:
[image]
bug is also on other windows machine( not Michael specific) I will try on a mac when available
I also reproduce it slightly different as in this print-screen.
to make it appear it also a role of playing only with the event in second line, but scroll down and up it appears only first time until you scroll down and up again.

stephanemagnenat, 2015-03-18:
Ok, it worked on Moti's computer. Could you try to write down the specifications (Windows version, graphic card, etc.) of the computer you test and in which case the bug appears and in which case not? Thanks!

motib, 2015-03-18:
When I said OK previously, I only tried out the original problem.

Now, I followed Michael's instructions from "20 hours ago" - except that I had to add a few lines so that the "third" line was not visible unless scrolled. There were no problems and the red line looked OK when I scrolled up.

mbonani, 2015-03-18:
I have reproduce it on a mac. Fanny will send you the image when she has
time.
Try really to play with the scroll up and down and it's appear sometime.

Michael Bonani

Excuse my brevity sent from my smartphone

stephanemagnenat, 2015-03-18:
Ok I managed to reproduce it, so it is likely a bug on our side if it appears on all platforms.

motib, 2015-03-18:
I managed to reproduce it. It didn't appear on following the initial instructions, but when I clicked the second line back and forth between the same event and different events, it occurred.

stephanemagnenat, 2015-03-18:
@mbonani could you try to add a "update();" at line 520 of `clients/studio/p/uginsThymioVPL/Scene.cpp` and tell me if this fixes the issue?

stephanemagnenat, 2015-03-18:
I tried a work around for this issue in commit e12f2af.

@mbonani please test the latest version with this fix in.

mbonani, 2015-03-18:
works for me with latest commits

stephanemagnenat, 2015-03-18:
Ok, I will close for now. Feel free to re-open if the problem re-appears.

closed 2015-03-18 by stephanemagnenat