aseba-community/aseba #361: Replace or delete action blocks

motib, 2015-02-18:
Currently, if there is an action block within a frame, you can only add new blocks. I would like to be able to replace or delete a block. (If delete is implemented, replace is not really needed.) The Scratch interface seems appropriate: if you drag a block outside the frame, it is deleted.

stephanemagnenat, 2015-02-18:
I agree. There was a technical difficulty in implementing that, but I think that we should have a second look.

motib, 2015-03-18:
This is (correctly) labeled a "Wish".
I don't see it as an issue that should block the release of 1.4.

stephanemagnenat, 2015-03-18:
I have added a button to delete actions in commit 590e24a.

motib, 2015-03-19:
I assume that this is not in build 842463b. Could you please run the Windows build so that I can try this out?

stephanemagnenat, 2015-03-19:
I do not know why the build farm did not run last night, I have asked EPFL people to have a look.

mbonani, 2015-03-19:
Last built online

motib, 2015-03-19:
Fantastique! This will save me a lot of clicking!

Bug?: When you delete an event or action, no error message occurs.

I think there might be some confusion because the identical "x" symbol is used both for the blocks and for the lines, but I don't have a good alternative suggestion: A different color would be distracting and a different symbol would conflict with the well-known meaning of "x" for deletion.

stephanemagnenat, 2015-03-19:
@mbonani do we know why it did not run automatically?

stephanemagnenat, 2015-03-19:
The drop out of the area is missing, but I do not want to delay 1.4 release for that. So I update the bug and re-target milestone 1.5

mbonani, 2015-03-19:
yes, VM is on a personal computer (Dany). Thecomputer crash reboot but the
VM making the packet was not relauch. We are looking to put on a school
server but don't still get answer...

*Bonani Michael*
Tél.: +41 (0) 21 693 78 69
Mobile: +41 (0) 76 349 72 14

Association Mobsya
1023 Crissier
www.mobsya.org

stephanemagnenat, 2015-03-19:
Thanks for the explanation! At some point (at the start of the KTI?) we should find a solution to build Aseba for several platform every night. In the mean time, a school computer is a good idea.

mbonani, 2015-03-23:
This is the plan yes, waiting CTI to make a build server on a mac mini, people will have more competence to mount it at this time.

stephanemagnenat, 2018-02-02:
As no further work beside bug fixing is planned on VPL1, I am removing this issue from the milestone.