aseba-community/aseba #362: Default file name of screenshot

motib, 2015-02-18:
The default file name should be the same as that of the VPL source file with a graphics extension (png). Currently, the last file name is used.

stephanemagnenat, 2015-02-18:
I see your point, although one can argue that the last file name is better, for instance if one saves several times an image, while varying only part of the name, such as a trailing number for example.

motib, 2015-02-18:
Since the screenshots are of the VPL source code _only_, I can't see why you would want multiple images. When I have several versions of a program, I naturally give them different names using trailing numbers (prog1.aesl, prog2.aesl, ...) so that I can save them all. It follows that the corresponding image files will also have different names (prog1.png, prog2.png, ...).

stephanemagnenat, 2015-02-18:
That is a good point indeed.