aseba-community/aseba #364: Enable feedback by default

stephanemagnenat, 2015-03-02:
The research student conducted in the context of the ITiCSE 2015 paper shows that visual feedback of the executed events improves the learning. We should enable it by default for VPL 1.4, but make sure that does not create any problem.

Please test and report problems here.

mbonani, 2015-03-16:
Could this be an option? Or configurable inside the program instead at start-up?

stephanemagnenat, 2015-03-17:
I think that if there are no technical reason to have it enabled, I would prefer to have it always: the more configurability we add the more complex it becomes to the user.

mbonani, 2015-03-17:
I see two negative points,  that are to be balanced:
- on recursive events the blinking is not nice and it seems that there is a bug...
- it's add line of code that perhaps confuse the user

stephanemagnenat, 2015-03-17:
I agree about the line, although we can make it shorter if we just want to blink and not to display the whole event. In addition, I can make a small editing pass to hide it, although maybe we do not want to hide it for educational reasons.

We could add a small "anti-bounce" behaviour to avoid excessive blinking in case of repetitive events.

What do you think?

stephanemagnenat, 2015-03-17:
Implementing feedback by default makes sense only if issue #336 is solved or worked around

motib, 2015-03-18:
I don't understand the connection with 336 which is closed ...

Enabling feedback with a command-line switch means that it will probably never be used by an ordinary teacher or student. It is just too complicated to explain how to change the "Properties" of an icon (at least in Windows). Feedback should either be an option at runtime or an unchangeable default. I support the latter: there are _many_ design decisions in VPL and this is just one more of them. I see no reason to complicate the interface with an "options" icon. In fact, there is a precedent: removing the color selection option.

stephanemagnenat, 2015-03-18:
Hi, sorry I meant #366, I have corrected.

I agree that a command-line switch means that no one will use it, and that is the reason why I want to enable it by default, given its strong effect on improving learning. The problem is that currently, on OS X, with feedback enabled, when one reconnects the robot there is quite a chance of receiving corrupted data which will make Thymio VPL crash due to consistency checking within Aseba's packet de-serialization system.

stephanemagnenat, 2016-01-26:
Fully implemented both in Studio and VPL standalone and enabled by default in commit 6950d8a.

closed 2016-01-26 by stephanemagnenat