aseba-community/aseba #366: Reconnection does not always work on OS X with Thymio II

stephanemagnenat, 2015-03-17:
On OS X, reconnection does not always work as it sometimes leads to corrupted packets. This is surprising as the firmware of Thymio II properly handles disconnections, including during packet send and reception.

This bug is very easy to reproduce. In Studio while connected to a Thymio with USB, create a global event named `Test` and input this simple program:
```timer.period[0] = 50
onevent timer0
emit Test [0,0,0,0,0,0,0,0,0,0]```
This program has a low bandwidth, of about 500 bytes per second.

With OS X, Studio will receive corrupted packets with a high probability after reconnection.

Tested on OS X 10.10.2

motib, 2015-03-18:
Trying on Windows 7. Exactly what is supposed to go wrong?!
I disconnected and reconnected, clicked Load and Run and it executed OK.

stephanemagnenat, 2015-03-18:
> Exactly what is supposed to go wrong?!

On OS X, Studio crashes on reconnection, because it receives corrupted data. I do not know why.

riedo, 2015-03-19:
I cannot reproduce on OSX 10.7.5.
where was your version compiled? is it mine or a homebrew one?
Does the Thymio reconnect correctly if there is no event?

I had cases of crash on reconnection when compiling with 10.6 sdk and running on 10.5 for example... could it be similar?

stephanemagnenat, 2015-03-19:
> I cannot reproduce on OSX 10.7.5.

Very interesting.

It is an homebrew one. I was using the system-provided compiler through XCode, which is clang on 10.10.2 (Apple LLVM version 6.0 (clang-600.0.57) (based on LLVM 3.5svn)).

Thymio reconnects correctly if there is no event, and the probability of crash increases with the bandwidth.

It would be interesting to test my builds on another system, who knows, maybe the problem is with the serial driver of 10.10. Which version of OS X do you have?

riedo, 2015-03-19:
As I said, 10.7.5 :)
can you reproduce the bug with my build on your computer?

stephanemagnenat, 2015-03-19:
I confirm that I could not reproduce the bug with your build.

riedo, 2015-03-19:
In this case I get a wrong Test event followed by two Toto events, and then back to normal.
Interestingly, always the same sequence of corrupted events.
Why would this depend on the build? I must say that if this is some kind of serial driver problem, I lack skills to solve it.
[image]

stephanemagnenat, 2015-03-19:
I do not know why this depends on the build, it is weird. To solve it, a first step would be to launch the program with `dtruce` (`strace` equivalent on OS X) to see what we read from the serious port (to rule out a bug in Dashel). We could see whether your build or my build read different data. If the data read from the kernel are already corrupted, then we would need to use a USB logic analyser to see if data are lost in the kernel somewhere. René Beuchat at EPFL/HEIG has the skill to do this, but I do not know if he is knowledgable with `strace`-like tools.

For a start, I will test on Linux, as Linux and OS X share the same Dashel code.

stephanemagnenat, 2015-04-15:
Interestingly, the broken data do not appear directly upon reconnection, but somewhat later. Using the provided test case, close Studio, starting a switch, disconnecting the robot, closing and re-starting the switch, the following error appears:
```
0x1 user message from 1 : user message of size 10 :    0    0    0    0    0    0    0    0    0    0 
 0x1 user message from 1 : user message of size 10 :    0    0    0    0    0    0    0    0    0    0 
 0x1 user message from 1 : user message of size 10 :    0    0    0   20    1    1    0    0    0    0 
   0 user message from 0 : user message of size 0 : 
   0 user message from 0 : user message of size 0 : 
 0x1 user message from 1 : user message of size 10 :    0    0    0    0    0    0    0    0    0    0 
 0x1 user message from 1 : user message of size 10 :    0    0    0    0    0    0    0    0    0    0 
 0x1 user message from 1 : user message of size 10 :    0    0    0    0    0    0    0    0    0    0 
```

stephanemagnenat, 2015-04-15:
An hypothesis could be that when re-opening the serial port on OS X after a disconnection, the FIFO of the driver still have old data, including a non-completed packet.

stephanemagnenat, 2015-04-15:
This hypothesis does not seem to old with the following test:

Using that program:
```
var i = 0

timer.period[0] = 50
onevent timer0
emit Test [i,0,0,0,0,0,0,0,0,0]
i += 1
```

Launching Studio, stopping Studio, launching switch and disconnecting:
```
 0x1 user message from 1 : user message of size 10 : 2880    0    0    0    0    0    0    0    0    0 
 0x1 user message from 1 : user message of size 10 : 2881    0    0    0    0    0    0    0    0    0 
 0x1 user message from 1 : user message of size 10 : 2882    0    0    0    0    0    0    0    0    0 
[Wed Apr 15 10:17:52 2015 732] Abnormal connection closed to ser:name=Thymio-II;baud=115200;stop=1;parity=none;fc=none;bits=8 : File read I/O error. Device not configured
```
Waiting some seconds, then reconnecting and re-launching switch:
```
0x1 user message from 1 : user message of size 10 : 3019    0    0    0    0    0    0    0    0    0 
 0x1 user message from 1 : user message of size 10 : 3020    0    0    0    0    0    0    0    0    0 
 0x1 user message from 1 : user message of size 10 : 3021    0    0   20    1    1 3022    0    0    0 
   0 user message from 0 : user message of size 0 : 
   0 user message from 0 : user message of size 0 : 
 0x1 user message from 1 : user message of size 10 : 3023    0    0    0    0    0    0    0    0    0 
 0x1 user message from 1 : user message of size 10 : 3024    0    0    0    0    0    0    0    0    0 
 0x1 user message from 1 : user message of size 10 : 3025    0    0    0    0    0    0    0    0    0
```

This shows that disconnection happened at packet 2882. Packet 3019 was received upon reconnection, and was ok. But packet 3021 was not complete.

mbonani, 2015-04-15:
Test on windows nothing similar have been seen on firmware 8 and 9.1
But automatic reconnection not work the same. It is not possible on switch (have to relaunch switch) and on studio you have to have "auto refresh" enable to go into the reconnection mode. But at reconnection no problem of such wrong packets...

stephanemagnenat, 2015-04-16:
I tried on Linux (kernel 3.6.16, Ubuntu 14.04 x86-64) and could not reproduce the problem we see on OS X.

stephanemagnenat, 2015-04-16:
Interestingly, on Linux with firmware 9 I get a wrong-sized message on reconnection sometimes. Maybe this also happens with firmware 8. I am confused, could it be that the mechanism to actually detect connection on Thymio (DTE) is not working?

stephanemagnenat, 2015-04-16:
Testing with a modified firmware that turns a wheel when the Thymio thinks the USB connection is opened, we see that the latter is opened upon physical reconnection on OS X. On Windows, the wheel turns a little upon reconnection and then stops. Therefore, it seems that the DTE detection is an unreliable way to know if we are connected.

stephanemagnenat, 2015-04-16:
Commit b131546 proposes a work-around that corrects the issue here on OS X (use latest Dashel). I need to test on other POSIX platforms such as Linux before assuming this bug fixed.

stephanemagnenat, 2015-04-16:
I still managed to get a broken packet with this fix, using firmware 8. So more testing is needed, both with firmware 8 and latest 9.1.

stephanemagnenat, 2015-04-16:
I got a broken packet using ThymioVPL on OS X, despite the correction.

stephanemagnenat, 2015-04-16:
Commit 1f57f66 attempts to make re-connection more robust, hopefully correcting this problem.

riedo, 2015-04-16:
on OSX10.7.5, with thymio firmware 9.1, aseba & dashel recompiled on april 16, we could not make aseba crash or trigger kernel panics anymore.
We stil got corrupted data however (hard to reproduce though)

[image]

and with firmware 8 studio crashes

stephanemagnenat, 2015-04-16:
Corrupted data is as bad as crash. Do you have any indication of the situation these happened? I guess you compiled Aseba with commit 1f57f661d3ee28ef6cf255fd1c673e2604053b9b? Of course timing issue can still lead to some problems, but these should now very very unlikely. If you got a problem in minutes of testing, I would say that there is still some cases we are missing.

riedo, 2015-04-16:
I do not agree that corrupted data every few minutes is as bad as systematic crash with frequent kernel panic. However the point was to give a precise description of the bug  and what happens (instead of "not working") and not a judgement of value.

Yes, that's the commit I used.

I still use the same test case.
```timer.period[0] = 50```
```onevent timer0```
```emit Test [0,0,0,0,0,0,0,0,0,0]```

with events:
toto (no args)
Test (10 args)

stephanemagnenat, 2015-04-16:
> I do not agree that corrupted data every few minutes is as bad as systematic crash with frequent kernel panic.

A kernel panic no indeed. But the fact that Aseba can trigger a kernel panic just speaks about bad implementation of OS X and no much more.

On the side of corrupted data vs Aseba crash, actually Aseba does not crash because there is a bug in it, but because there are some consistency checking in the message de-serialization system that abort Aseba when impossible condition happens on messages. So having these assertion or wrong data is exactly similar, the fact that one or the other happens depends on the precise content of the messages. So I want to hunt down all data corruption.

stephanemagnenat, 2015-04-20:
Further tests (disconnect on a computer and reconnect on another one) show that the problem is indeed in the Thymio's firmware, in which DTR is not cleared upon disconnection. I will keep this issue open until the firmware's issue is fixed.

riedo, 2015-05-05:
So, I tested with a longer dely in studio as Stephane suggested
Firmware modified today 5 may 2015 by Michael (supposed to clear DTR, and on reconnection start turning 1 wheel when the port opens, start turning the second wheel after the delay when Thymio starts sending packets)
Wheels turned when I started Studio, and stopped if I disconnected the robot.
On reconnection, nothing moved for a long time at first, then 1st wheel startd, then 2nd wheel started with aseba showing events again. According to Michael this is the expected behaviour.

Then, with the same test case as before (test event with 10 args every 50 ms)
Aseba commit: 1aa42de
Dashel commit: e88c118
at the ~15th disconnection, the wheels did not stop turning as the should have, and upon reconnection there was a broken packet.
[image]

stephanemagnenat, 2015-05-05:
> at the ~15th disconnection, the wheels did not stop turning as the should have, and upon reconnection there was a broken packet.

Ok so there is still a bug in the firmware.

riedo, 2015-05-05:
more bugs description (I continued the test from above)
I had 4 more cases where the wheels did not stop turning but there were no broken packets.
I had one case where the wheels stopped, but Aseba crashed on reconnection, I'm just surprised because the bug report seems different than usual; is it also because of broken packets? Looks like segfault, not abort, if i'm not mistaken.

```
Process:         asebastudio [1433]
Path:            /Users/USER/*/Aseba Studio.app/Contents/MacOS/asebastudio
Identifier:      org.mobsya.asebastudio
Version:         ??? (1.2)
Code Type:       X86-64 (Native)
Parent Process:  launchd [133]

Date/Time:       2015-05-05 11:01:42.534 +0200
OS Version:      Mac OS X 10.7.5 (11G63)
Report Version:  9

Interval Since Last Report:          11670324 sec
Crashes Since Last Report:           294
Per-App Interval Since Last Report:  318542 sec
Per-App Crashes Since Last Report:   34
Anonymous UUID:                      2A8DEFDF-6C37-4B80-B7C0-C73C5353D1AC

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: KERN_INVALID_ADDRESS at 0x0000000000000010

VM Regions Near 0x10:
--> 
    __TEXT                 0000000100000000-0000000100484000 [ 4624K] r-x/rwx SM=COW  /Users/USER/*/Aseba Studio.app/Contents/MacOS/asebastudio

Application Specific Information:
objc[1433]: garbage collection is OFF

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.mobsya.asebastudio        	0x0000000100026658 Aseba::NodeTab::processCompilationResult(Aseba::NodeTab::CompilationResult*) + 1304
1   org.mobsya.asebastudio        	0x0000000100048175 Aseba::NodeTab::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) + 197
2   QtCore                        	0x000000010161e64e QMetaObject::activate(QObject*, QMetaObject const*, int, void**) + 1566
3   QtCore                        	0x00000001014da947 QFutureWatcherBase::event(QEvent*) + 119
4   QtGui                         	0x00000001006b62ad QApplicationPrivate::notify_helper(QObject*, QEvent*) + 189
5   QtGui                         	0x00000001006bce04 QApplication::notify(QObject*, QEvent*) + 1060
6   QtCore                        	0x00000001016061ac QCoreApplication::notifyInternal(QObject*, QEvent*) + 124
7   QtCore                        	0x00000001016075d0 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 736
8   com.apple.CoreFoundation      	0x00007fff90cd04f1 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
9   com.apple.CoreFoundation      	0x00007fff90ccfd5d __CFRunLoopDoSources0 + 253
10  com.apple.CoreFoundation      	0x00007fff90cf6b49 __CFRunLoopRun + 905
11  com.apple.CoreFoundation      	0x00007fff90cf6486 CFRunLoopRunSpecific + 230
12  com.apple.HIToolbox           	0x00007fff8ddb12bf RunCurrentEventLoopInMode + 277
13  com.apple.HIToolbox           	0x00007fff8ddb84bf ReceiveNextEventCommon + 181
14  com.apple.HIToolbox           	0x00007fff8ddb83fa BlockUntilNextEventMatchingListInMode + 62
15  com.apple.AppKit              	0x00007fff93d17779 _DPSNextEvent + 659
16  com.apple.AppKit              	0x00007fff93d1707d -[NSApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 135
17  com.apple.AppKit              	0x00007fff93fcb735 -[NSApplication _realDoModalLoop:peek:] + 610
18  com.apple.AppKit              	0x00007fff93fc4f4c -[NSApplication runModalSession:] + 48
19  QtGui                         	0x000000010066e78e QEventDispatcherMac::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 1422
20  QtCore                        	0x00000001016050c4 QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 68
21  QtCore                        	0x0000000101605474 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 324
22  QtGui                         	0x0000000100b5f7c5 QDialog::exec() + 229
23  org.mobsya.asebastudio        	0x0000000100081448 Aseba::DashelTarget::disconnectionFromDashel() + 488
24  QtCore                        	0x000000010161a741 QObject::event(QEvent*) + 1073
25  QtGui                         	0x00000001006b62ad QApplicationPrivate::notify_helper(QObject*, QEvent*) + 189
26  QtGui                         	0x00000001006bce04 QApplication::notify(QObject*, QEvent*) + 1060
27  QtCore                        	0x00000001016061ac QCoreApplication::notifyInternal(QObject*, QEvent*) + 124
28  QtCore                        	0x00000001016075d0 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 736
29  com.apple.CoreFoundation      	0x00007fff90cd04f1 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
30  com.apple.CoreFoundation      	0x00007fff90ccfd5d __CFRunLoopDoSources0 + 253
31  com.apple.CoreFoundation      	0x00007fff90cf6b49 __CFRunLoopRun + 905
32  com.apple.CoreFoundation      	0x00007fff90cf6486 CFRunLoopRunSpecific + 230
33  com.apple.HIToolbox           	0x00007fff8ddb12bf RunCurrentEventLoopInMode + 277
34  com.apple.HIToolbox           	0x00007fff8ddb856d ReceiveNextEventCommon + 355
[... 281 lines cut ...]
===========                      =======
TOTAL                             470.1M

Model: MacBookPro8,2, BootROM MBP81.0047.B27, 4 processors, Intel Core i7, 2.3 GHz, 8 GB, SMC 1.69f4
Graphics: AMD Radeon HD 6750M, AMD Radeon HD 6750M, PCIe, 1024 MB
Graphics: Intel HD Graphics 3000, Intel HD Graphics 3000, Built-In, 512 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1333 MHz, 0x80AD, 0x484D54333531533642465238432D48392020
[... cut ...]
USB Device: BRCM2070 Hub, 0x0a5c  (Broadcom Corp.), 0x4500, 0xfa110000 / 5
USB Device: Bluetooth USB Host Controller, apple_vendor_id, 0x821a, 0xfa113000 / 8
USB Device: Apple Internal Keyboard / Trackpad, apple_vendor_id, 0x0253, 0xfa120000 / 4
USB Device: FaceTime HD Camera (Built-in), apple_vendor_id, 0x8509, 0xfa200000 / 2
USB Device: hub_device, 0x0424  (SMSC), 0x2513, 0xfd100000 / 2
USB Device: USB Optical Mouse, 0x0461  (Primax Electronics), 0x4d22, 0xfd120000 / 4
USB Device: IR Receiver, apple_vendor_id, 0x8242, 0xfd110000 / 3

```

stephanemagnenat, 2015-05-05:
> I had 4 more cases where the wheels did not stop turning but there were no broken packets.

That can happen, as packets are not sent continuously. That still means the bug in the firmware showed up.

> I had one case where the wheels stopped, but Aseba crashed on reconnection, I'm just surprised because the bug report seems different than usual; is it also because of broken packets? Looks like segfault, not abort, if i'm not mistaken.

I would say so, as the abort is just a sanity check about the evenness of packets, but if packets are broken, this check can pass (actually by chance it would pass half of the time) but inconsistent messages can be created and Aseba might crash further.

riedo, 2015-05-05:
I see. So in that case the fact that the wheels stopped turning on disconnection did not indicate correct behaviour either.

mbonani, 2015-05-05:
In firmware I add in the usb_uart_tick() function the line to clear DTR when disconnected
````
// Call this regularly, it manage the connection/disconnection of usb and charge enable
// It's absolutly not time-critical
void usb_uart_tick(void)
{   
	int vbus = U1OTGSTATbits.SESVD;
// --- Attaching/detaching part
	// If we are detached and we sens vbus, then attach
	if(vbus && (USBGetDeviceState() == DETACHED_STATE))
	{
		USBDeviceAttach();
	}
	
	// If we are not detached and we don't sens vbus, then detach 
	if(!vbus)
	{
		USBDeviceDetach();
                control_signal_bitmap.DTE_PRESENT=0; //clean DTR/DTE
	}
```
This functions is call by update_aseba_variables_read():
```
void update_aseba_variables_read(void)
 {
	// TODO: REMOVE ME (move to behavior ? /!\ behavior == IPL 1 !! race wrt aseba !)
	usb_uart_tick();

	motor_get_vind((int *) vmVariables.uind);
}
```
itself call either by idle_without_aseba() and run_aseba_main_loop() that are in the main loop...

For me having the robot in  few case continuing thinking having connection could have a problem in the main loop of aseba... could we perhaps try to move the usb_uart_tick() else where but I am not sure to understand this comment:
 `// TODO: REMOVE ME (move to behavior ? /!\ behavior == IPL 1 !! race wrt aseba !) `

stephanemagnenat, 2015-05-05:
> So in that case the fact that the wheels stopped turning on disconnection did not indicate correct behaviour either.

If wheels continue to run once disconnected, there is certainly a bug. If wheels stop, it says nothing on whether there is a bug or not. If wheels start again on reconnection, but not port opening, there is still the bug. That is why I suggested a large reconnection delay on Aseba, to test that.

stephanemagnenat, 2015-05-05:
@mbonani I do not have time to look in the firmware now (I will try in the next days), I suggest that you email @retornaz for review of the DTR fix.

stephanemagnenat, 2015-05-05:
@mbonani without going into the details of the firmware, you patch looks correct (the comment should be "clear DTR/DTE" and maybe the indentation is wrong). Maybe one thing to check: are you sure that the bit must be put to 0 to clear DTR? It might be active-low (even though reading the Wikipedia page on DTR seems to indicate that 0 is indeed the right value, but who knows maybe the API of the dspic is weird). We probably need the light of @retornaz here.

stephanemagnenat, 2015-05-28:
This seems to be fixed in firmware 9 (see aseba-community/aseba-target-thymio2#5)

closed 2015-05-28 by stephanemagnenat