aseba-community/aseba #367: Inconsistent display empty e-a pair

motib, 2015-03-24:
There is an inconsistency in the appearance of an empty e-a pair. Initially and when a new empty pair is displayed automatically, an empty frame appears. When + is clicked, the empty frame includes subframes for the event and action, and, furthermore, it is colored yellow to indicate that it is the current frame. I suggest that the subframes always be displayed and that current frame always be indicated by the yellow background.

stephanemagnenat, 2015-03-24:
The reason of this seemingly inconsistent behaviour is the combination of two desiderata:
- A way to indicate to users where they can extend the program in a lightweight way (the dotted frame)
- The requirement to give the users control how they build the program, i.e. to add a certain number of empty pairs and then fill them.

Regarding the yellow background, this means that a frame is selected. A selected frame has its code highlighted, and its event replaced when one clicks the left event pane, and an action added when one clicks the right action pane. The behaviour is the same metaphor as a selected text in a text editor, even though the modality of selection (click in VPL and click-drag in a text editor) is different.

motib, 2015-03-24:
I have a teacher working with me (Asaf) and I asked him to review the current version of the VPL tutorial. He noticed inconsistencies and I found it hard to write understandable explanations :-(.

> extend the program in a lightweight way

A solution is not "lightweight" if there are _two_ graphical representations that really mean the same thing, because it imposes an additional cognitive load. The explanation of the current situation is not at all easy: "you get a simple frame initially and automatically, and subframes appear once you drag in a block, but if you click + you get a frame with subframes even before you drag something in". This seems ad hoc and I don't know how to explain why the behavior should be different.

> add a certain number of empty pairs and then fill them.

I don't think this is important, but let's say that it is. The current implementation is inconsistent because you can't create more than one empty pair initially. Furthermore, how do I explain that the first n-1 empty pairs have subframes and the n'th doesn't??

> A selected frame has its code highlighted

Great. Asaf even found what I didn't know before: you can simply click a block and don't have to drag and drop it! But I don't think that selected text is the right metaphor. It would be more appropriate to use the metaphor of a cursor. (Text / a block) is entered (at / in) the current (position of the cursor / highlighted pair). Again, the problem is inconsistency: (1) initially, there is no highlighted frame but you can insert blocks, although the frame is _not_ highlighted; (2) you can insert a block into a non-initial non-highlighted simple frame and it _is_ highlighted; (3) a frame created with + is highlighted.

This is not a serious issue and in fact I didn't pay attention to it during all this time that I have working with VPL, but it is potentially confusing to students and certainly difficult to explain should a student ask!

stephanemagnenat, 2015-03-24:
I agree with your comments. The current (complicated) behaviour is the result of several compromises. I will think about it a bit more and also ask the designers for their impression, and then see whether changes are needed. If so, I will evaluate whether they should go into 1.4 or later.

motib, 2015-03-24:
OK!

stephanemagnenat, 2015-03-24:
I have tried a new behaviour in commit 760a6f15db21badca7e206aa981ec555605ac55d, please test.

stephanemagnenat, 2015-03-24:
About highlighting: dropping a block will highlight the frame, clicking a block will not. The reason is that these two ways of adding a block come from different use cases: dropping a block on a line will likely leads to editing this line, while clicking would likely be part of clicking in a sequence several event blocks, and then several action blocks. I agree that it is a bit complicated to explain.

Furthermore, in commit 760a6f15db21badca7e206aa981ec555605ac55d the highlighting of an empty line is disabled, but the underlying mechanism is not disabled, so clicking + will add a new selected line, and therefore clicking on a block will add it to that line. It feels natural at usage but I do not like the idea that the selection is not visible any more.

motib, 2015-03-25:
Thanks for removing the highlighted empty line!!
That will remove a confusing footnote from the tutorial :-).

I'm not concerned about the semantics of clicking a block, because drag-and-drop is well known and easy to learn.

stephanemagnenat, 2015-03-25:
So you are happy with the current behaviour?

motib, 2015-03-25:
I'm happy with the consistency in the display of the empty lines. I still don't quite understand all the subtleties of clicking blocks to enter them in a program, but since I never do that it doesn't bother me :-).

I can close the issue, unless you want to leave it open for later consideration about clicking blocks.

stephanemagnenat, 2015-03-25:
I would like to keep it open until the designer gave their feedback.

beltranmaria, 2015-03-25:
As far as I remember it was also something that I found weird, mostly because if by mistake you didn’t want to take a given block and you just realise it so you drop the cursor then it will be like a click and it will anyways put it inside the program… or people will just click because they move the mouse a bit etc. I think it can lead to unwanted actions...
_____________________
MARIA  MARIA BELTRAN
Product & Interaction Designer
www.apelab.ch <http://www.apelab.ch/>
www.mariamari-a.com <http://www.mariamari-a.com/>

stephanemagnenat, 2015-03-26:
After discussing with the Maria (the designer), in commit abf876be9554f6ac66c02a3c4055bdb6f447e17d I implemented your initial suggestion, please test.

motib, 2015-03-26:
If you'll lend me a Thymio, I can test it when I get to Lausanne :-)).

closed 2015-03-31 by stephanemagnenat

motib, 2015-04-08:
Checking build: 02 April 2015 01:17 f8491a6.

I'm still having problems understanding the yellow highlight: (1) the initial blank line is not highlighted; (2) if you click on a line to highlight it and then delete it, no line is highlighted; (3) there is a different semantics when clicking on an event block depending on whether the highlight exists (replace) or not (drop in next line); (4) clicking on the gray background of a line highlights it, but clicking on a block doesn't.

As I wrote above, if the semantics of highlighting is the same as that of a cursor indicating the "current block", it should all be consistent and I think easy to understand and explain.

Stephane: I'll let you decide whether to re-open the issue. Alternatively, you can write an explanation of the behavior  :-)).

stephanemagnenat, 2015-04-14:
> (1) the initial blank line is not highlighted;

Yes that is on purpose (so that one can click events and actions to add several new lines). In which respect is that a problem?

> (2) if you click on a line to highlight it and then delete it, no line is highlighted

That is true, selection is optional, one does not have to select a line, one can.

> (3) there is a different semantics when clicking on an event block depending on whether the highlight exists (replace) or not (drop in next line)

Yes, that is on purpose.

> 4) clicking on the gray background of a line highlights it, but clicking on a block doesn't

That is inconsistent, I did not realise it, and will gladly correct it.

> As I wrote above, if the semantics of highlighting is the same as that of a cursor indicating the "current block", it should all be consistent and I think easy to understand and explain.

The semantic is the one of an optional current block: there might be 0 or 1 "current" block in the program.

motib, 2015-04-14:
The semantic is the one of an optional current block

I find this extremely non-intuitive, and others might agree if they have experience with word processors, etc., that use cursors to indicate the "non-optional" current location.

May I suggest that you add an explanation of the concept to the "Environment overview" page?!!

stephanemagnenat, 2015-04-15:
> I find this extremely non-intuitive, and others might agree if they have experience with word processors, etc., that use cursors to indicate the "non-optional" current location.

I do not fully agree. For instance in this github text editor, the cursor can be active, and the field editable, or disabled, and the field gray (if you click outside the editor). This semantics is not completely "alien" to users. But I agree that it might be too complicated for beginners. The thing is that I find very useful the behaviour of clicking on event and action blocks when no pair is selected. But maybe it is only me, and I should add a command line switch ;-)

> May I suggest that you add an explanation of the concept to the "Environment overview" page?!!

Ok, I will do so once we definitely agree on the behaviour.

motib, 2015-04-15:
I see what you mean by editable/disabled fields. It's like any form you fill in from a browser and so it is not "alien" at all. But the VPL interface is that of a simple editor like Notepad with only _one_ "field" in which data is entered. I don't see the point of having that single field sometimes editable and sometimes disabled.

I'd like to suggest something even more radical :-). Remove the click-to-enter and support just drag-and-drop. I don't think that total novices need more than one way of doing something. Scratch has 6,272,748 registered users and I never heard a complaint that it supports only drag-and-drop.

Whatever you do, don't add a command-line switch! No kid will use commands and it is too hard to explain how to modify the "Properties" of an icon (in Windows, at least).

stephanemagnenat, 2015-04-15:
> Remove the click-to-enter and support just drag-and-drop

On some Windows computer with buggy graphics drivers (the ones in ETH's computer room and Michael's one), the display of drag and drop does not work well, and the ability to click is sometimes helpful. But that aside, your suggestion also makes sense.