aseba-community/aseba #370: Add a button to control the volume of Thymio

cbarraud, 2015-04-02:
I wondered if we could place in Studio (or even in VPL) a button (or slider + button) to control the volume of Thymio.

The idea would be to have either a "Mute" and a "Volume on" button or a slider (with 8 positions) and a button "Save into Thymio".

The reason of this wish is coming from a lot of people (especially teachers and an ergonomics designer) that are annoyed by the sound of Thymio. In a class room with 10-20 robots, it's getting really noisy.

Technically, it would "just" request a flash into Thymio of some settings. I'm not sure it would be hard to implement. It could even be possible for 1.4... ?

mbonani, 2015-04-02:
The settings to change is the number 0 and value go from 0 to 8 (shift output of sound 0 means 100%, 1 50%, 2 25% etc....) but this functionality have to not flash each time the Thymio, it has to be a separate button that make it with the confirmation of the user (we have to keep the flash memory it has not infinite cycle)

stephanemagnenat, 2015-04-07:
As we discussed in real life, I suggest to have that in a "settings" behaviour in the Thymio, as this problematics also applies to the basic behaviour.

mbonani, 2015-04-10:
I think that both solution can be implemented. If somebody as his Thymio conneted to the computer, it is painful to unconnect enter in the setting mode (that not exist for now)  and make the change when it can be "easely" done in studio. Solution in studio is also working for all firmware for 2012

stephanemagnenat, 2015-04-10:
The thing is that I do not want to add feature creep. For Studio it is ok as it can be a plugin, but adding that to VPL seems a bit too specific.

stephanemagnenat, 2016-05-11:
Now that it is implemented in the firmware, I propose to close this request. Ok for you?

cbarraud, 2016-05-12:
Sure, I will close the issue.

closed 2016-05-12 by cbarraud