aseba-community/aseba #372: Red dots do not appear when updating

motib, 2015-04-19:
In the build from 15/4, when opening a vpl project from 1.3 or from earlier versions of 1.4, the red dots next to the white squares on the sensors do not appear. Workaround: click through the colors until the white appears again and the red dots will appear.

stephanemagnenat, 2015-04-20:
Could you please provide the offending file?

motib, 2015-04-20:
I can't attach from within this window but you can use _any_ of the files in the "programs14" directory.

In fact, the problem is more serious than I originally thought. Although the "workaround" displays the dots, when you save, click New to start over and re-open the file, they disappear again. A diff of the files before and after the workaround shows no difference between the two.

stephanemagnenat, 2015-04-20:
Corrected in commit f61f72f, thanks for testing and reporting!

closed 2015-04-20 by stephanemagnenat

motib, 2015-04-20:
I'll write the Guiness book of records that you fixed the bug in only 15 minutes!!
I hope that the windows build will run tonight....

motib, 2015-04-21:
OK now. Thanks.