aseba-community/aseba #376: Font size in Studio

motib, 2015-04-22:
I "wish" that there was an option to enlarge the font of (at least) the program area in Studio.

stephanemagnenat, 2015-12-01:
I have implemented this feature in commit 7ce536a, please test. Using ctrl and "+" or "-", or mouse wheel and "+"or "-", one can zoom in or out the aesl editor. If you are happy with the feature, I'll document it so that it is not a bug and close this issue :-)

mbonani, 2015-12-01:
zoom in and out works fine, but normal scrolling is now not working anymore.....

motib, 2015-12-01:
Could you please build me a Windows install so I can check it? Thanks.

stephanemagnenat, 2015-12-02:
Please take the latest master build from here: http://mobots.epfl.ch/aseba-nightly.php?show=wikidot

motib, 2015-12-16:
Works OK on Windows. Thanks.

I didn't encounter any problem with scrolling like Michael reported.

But ... it took me a while to figure out what to do - ctrl+scroll wheel on the mouse. Non-expert users probably won't think of this. I suggest that setting the font size be done (also?) in the View menu or View/Settings, so that it can be found easily and changed using a keyboard for people who have trouble
with the mouse. It would be sufficient to have Bigger font / Smaller font selections without implementing selection of an absolute size.

stephanemagnenat, 2015-12-16:
> I didn't encounter any problem with scrolling like Michael reported.

I corrected it already.

> But ... it took me a while to figure out what to do - ctrl+scroll wheel on the mouse. Non-expert users probably won't think of this. I suggest that setting the font size be done (also?) in the View [...]

The problem is that currently font size is attached to the editor, and not as a global property. Making it a global property is more, which is necessary for having a menu controlling it, is more work.

motib, 2015-12-16:
Would it be possible to implement this as + / - buttons on the keyword toolbar??

stephanemagnenat, 2015-12-16:
That would work, but it shocks my sense of consistency...

motib, 2015-12-16:
I was afraid you were going to say that :-( !
Would it sweeten the bitter medicine if you right justified "Font size [+] [-]" in the toolbar??

stephanemagnenat, 2016-01-21:
It should be implemented in commit 5f1016d through an entry in the edit menu. Please test with latest auto-generated build (http://mobots.epfl.ch/data/aseba-nightly/nightly-2016-01-21-aseba-bin-1.4.98-git-5f1016d-win32.exe) and close if you are happy with the result.

motib, 2016-01-21:
It is in the View menu, not the Edit menu; I assume that is what you meant.
Works great, but I would change the names to "Increase/Decrease font size".

stephanemagnenat, 2016-01-21:
I stole the terms from KDE, I assumed that they went through a lengthy argument to decide about it. But maybe enlarge is an unfortunate word...

motib, 2016-01-21:
As is "shrink" :-)

stephanemagnenat, 2016-01-21:
Changed in commit fe982e6

closed 2016-01-21 by stephanemagnenat