aseba-community/aseba #381: Cannot open files for wireless Thymio

motib, 2015-05-18:
In build nightly-2015-05-14-aseba-bin-1.3.9-git-d108090-win32.exe, the wireless Thymio is recognized for target selection, but then Studio gives an error message when attempting to open a file: "1 scripts have no corresponding node in the current network and have not been loaded." This occurs even with nodeIDs which used to work before.

stephanemagnenat, 2015-05-18:
Did you flash a new firmware into the Thymio? There was no change in Studio recently regarding to Thymio detection. If you send me the aesl file, I can check it here.

motib, 2015-05-18:
I didn't change the firmware.
It happens with any file (I sent a couple in a separate message).
But ... suddenly it started working again ... .

stephanemagnenat, 2015-05-18:
I tried both files you sent me and they all work. I suggest to close this bug, until we have more informations on which conditions make it happen.

closed 2015-05-18 by motib