aseba-community/aseba #386: Tap is much too sensitive in VPL

motib, 2015-06-22:
I am trying to simulate a Mindstorms touch sensor using the tap event in VPL. However, once the Thymio starts moving at any motor power greater than 200, the event occurs by itself. I guess you're trying to protect the robot from damage, but I think that the threshold should be such that the robot can move without a tap event occurring spontaneously.
Thanks
Moti

motib, 2015-06-22:
Followup: I looked at the values of the acc array in Studio and they don't seem to change when I tap the Thymio. Is the tap event in VPL caused by the microphone and not the accelerometers??? That would explain the above problem.

stephanemagnenat, 2015-06-22:
This needs testing indeed, although I do not have a Thymio with me right now. As this is only a constant change, we can still release it within 1.4

stephanemagnenat, 2015-06-29:
I think that the tap event is due to the accelerometers. I assume that the value for triggering this event is calculated in C code at a faster rate than the acc variables are updated. As far as I can see, the threshold for tap is stored in the hidden variable `acc._tap`. Maybe you can play with it and see if you can get a behaviour you like?

But it is clear that at a certain speed, the vibrations will be strong enough to generate the event, whatever the threshold.

marizafreire, 2015-06-29:
Bonjour,

Je ne suis pas disponible et serai de retour le 14 juillet. 

En cas de question urgente, merci de contacter christophe.barraud@mobsya.org ou michael.bonani@mobsya.org.

Meilleures salutations.

----------

Good afternoon,

I am not available and will be back on the 14th of July. 

In case of urgent matter, please contact christophe.barraud@mobsya.org or michael.bonani@mobsya.org.

Best regards.

motib, 2015-06-30:
I tried changing the value of acc._tap, but it changes back to the default 32, sometimes immediately and sometimes after one run, so I couldn't do extensive testing. However, it did seem that changing it to 100 resulted in better behavior.

The program I am using is:
onevent buttons
	if button.forward == 1 then
		motor.left.target = 250
		motor.right.target = 250
	end
onevent tap
	motor.left.target = 0
	motor.right.target = 0

retornaz, 2015-07-02:
FYI: the tap detection is done inside the accelerometer. 
The acc._tap variable is not the sensitivity but a flag used internally by the thymio firmware.

stephanemagnenat, 2015-07-14:
Ok, so there is no way to change the sensitivity without changing the firmware?

mbonani, 2015-07-14:
yes sensivitiy is hardcoded in the firmware. Moving at 150 is probably
sufficient...
I never paly with the value of tap detections and cannot say if it possible
to make tap detectiong not reconizing motor noise.  Probably Philippe as
choose them to be sufficient sensible to detect smal tap so that user don't
break the robot hiting it...

*Bonani Michael*
Tél.: +41 (0) 21 693 78 69
Mobile: +41 (0) 76 349 72 14

Association Mobsya
1023 Crissier
www.mobsya.org

stephanemagnenat, 2015-07-14:
Ok, so I understand that there is no simple solution and suggest to close the bug for now, ok?

motib, 2015-07-18:
I am disappointed that this feature is problematic. I would like to leave the issue open, so that the next round of firmware development will look into the issue, and see if some improvement can be done.

stephanemagnenat, 2015-07-22:
I agree. This issue depends on updating the firmware, I have created an issue on its repository: aseba-target-thymio2/issues#7. We can close this issue when that issue is corrected.

stephanemagnenat, 2015-08-13:
This cannot be solved with current hardware because this processing is done in the electronics of the accelerometer.

stephanemagnenat, 2015-11-24:
@mbonani could you please update this issue with the information about your latest firmware changes?

mbonani, 2015-11-24:
as explain in https://github.com/aseba-community/aseba-target-thymio2/issues/7
the tap event is now (firmware 10) less sensitive, specially when the robot is moving. You can now run at fullspeed (500) without firing the tap event.

closed 2015-11-24 by mbonani