aseba-community/aseba #387: VPL in Studio: advanced mode is reset when VPL is closed

stephanemagnenat, 2015-06-29:
When VPL is used in Studio in advanced mode, the advanced mode is lost when VPL is closed, and the generated text code corresponds to the simple mode.

stephanemagnenat, 2015-07-14:
Should be fixed in commit 72df17e, please test and close if solved in all cases.

motib, 2015-07-18:
I presume you mean that the code that appears in Studio is the same as that generated by the standalone VPL. If so, it seems OK.

riedo, 2015-07-20:
I post here because it might be a side effect of this update I can move if it is not the case.

Now when you close VPL the code stays, but if you try to edit it, it disappears (tested on Mac OS and Windows)

riedo, 2015-07-20:
It's actually worse. you cannot type any text at all any more. not even if you just start Aseba and want to type.
I will hide the mac version of the RC on the site for the moment.

stephanemagnenat, 2015-08-03:
I cannot reproduce that bug on Linux (but saw it on Michael's computer), which is bad news because I have no currently functional Windows or OS X debug environment.

I am slightly puzzled by this difference of behaviour between platforms, as I am by the possibility that commit 72df17e triggered this bug, although I do not say it did not.

I have some questions that might help me to fix this bug:
* Does the problem always appear, or only once the VPL window has been loaded showed once?
* If it does always occur, we should check whether this bug is really due to a change to VPL. To do so, one can comment the statement `reg("ThymioVisualProgramming", ASEBA_PID_THYMIO2, &createInstance<ThymioVPL::ThymioVisualProgramming>);` at line 106 of `clients/studio/PluginRegistry.cpp`. If the bug is still there, it is probably not due to commit 72df17e or to VPL.
* If the bug disappears, I suggest to check that the bug does does not appear at commit a48de90, and that it appears at commit 72df17e (the one after). If it is the case, I am really puzzled and need to add debug code at different places to understand what happens. If the bug is present in both commits, we need to go more into the past and bisect from there. If the bug is absent in both commits, we can bisect between 72df17e and head.

mbonani, 2015-08-03:
Bug is present even if you don't start VPL. 
I comment statement as you suggest and bug disappear.
Then I revert 72df17e and bug is also disapear. And original bug that you loose advance mode when closing stusio is not present for me.

stephanemagnenat, 2015-08-04:
So you say that as of commit 72df17e, this bug is not present, right? Commit 72df17e is indeed supposed to correct this bug, so it is normal that it is not present. @riedo, could you try commit 72df17e on OS X and tell me if the bug is present as of this commit? If not, the problem is with more recent commits. There are only 6 so it would be easy to bisect.

stephanemagnenat, 2015-08-04:
The new bug should be fixed in commit 32a9d51.

riedo, 2015-08-04:
seems good to me on mac OS: I can type text, create advanced programs in VPL and not lose them once I close the VPL window, and still edit them afterwards.

closed 2015-08-10 by riedo