aseba-community/aseba #389: No stack check for SUB_CALL and SUB_RET in vm.c

supcik, 2015-07-12:
There is no stack OVERFLOW/UNDERFLOW check for subroutine calls (vm.c, lines 493 - 511). Is there a good reason for not checking the stack there?

stephanemagnenat, 2015-07-13:
Yes, these are checked statically by the compiler, see `compiler/tree-emit.cpp` lines 111-135 for the computing of the depth, and `compiler/analysis.cpp` line 37 for the checking of the depth against the stack size of the target.

stephanemagnenat, 2015-07-13:
But I agree that to be consistent, these should be checked as well when `ASEBA_ASSERT` is defined.

This makes me think that out of consistency, we should also probably check every assignment to `vm->pc` to make sure it is not larger than the size of `vm->bytecode`. This could be done in every assignations to PC, but it would make the code larger, or it could be done just before deferencing `vm->bytecode

stephanemagnenat, 2015-07-13:
I am quite busy these days, and I do not want to do these changes in a rush, but if you want to provide a patch adding these asserts along with corresponding unit tests, I would be happy to review and merge it.

stephanemagnenat, 2015-11-12:
Should be fixed in commit acb0b87. Please check and close the issue if you think it is appropriate.

stephanemagnenat, 2016-04-21:
No answer, closing the issue.

closed 2016-04-21 by stephanemagnenat