aseba-community/aseba #390: VPL does not interpret pairs in the same order in basic and advanced mode

riedo, 2015-07-14:
A user had a code that behaved differently when switching to advanced mode, without changing anything else. On the screenshots we can see that even though the pairs are in the same order, the "call leds.top(0,0,0)" line is not at the same spot in basic and advanced mode, creating different behaviours.

[image]
[image]

stephanemagnenat, 2015-07-14:
Interesting. I will have to investigate.

The "bug" label is arguable, as we never specified an order of execution in case of race condition. But I agree that this change is inconvenient for the user.

riedo, 2015-07-14:
Yes. I remember Moti explained a similar "feature" in scratch when he came to lausanne.
Now this is your project, so you have the power to define what is a bug and what is a feature ;-) 

but i can assure you, users expect that a same sequence of order in a programming language generates the same bytecode. In my opinion it is not just a problem of execution which could illustrate race conditions and such, but a problem of consistency in the way the VPL code is translated to aseba script, making it difficult for users to understand.

stephanemagnenat, 2015-07-14:
I agree with you, and plan to investigate it, when time permits.

stephanemagnenat, 2015-07-14:
Fixed in commit 351d5f3, please test and close if it works for you.

riedo, 2015-07-14:
seems all good :) amazingly quick bugfix

closed 2015-07-14 by riedo

stephanemagnenat, 2015-07-14:
It was just a typo in the implementation actually, the idea was correct already. But it still took me some time to remember what the idea was ;-)