aseba-community/aseba #392: State update in VPL has disappeard

FrancescoMondada, 2015-07-15:
In the following code, generated by VPL, the update of the states at the end of the program has disappeared....

# variables for state
var state[4] = [0,0,0,0]
var new_state[4] = [0,0,0,0]

# reset outputs
call sound.system(-1)
call leds.top(0,0,0)
call leds.bottom.left(0,0,0)
call leds.bottom.right(0,0,0)
call leds.circle(0,0,0,0,0,0,0,0)

# subroutine to display the current state
sub display_state
	call leds.circle(0,state[1]*32,0,state[3]*32,0,state[2]*32,0,state[0]*32)

onevent prox
	when prox.horizontal[2] > 2000 do
		motor.left.target = -250
		motor.right.target = 300
	end
	when prox.ground.delta[1] < 400 do
		motor.left.target = -200
		motor.right.target = 350
		new_state[1] = 1
	end
	when prox.ground.delta[1] > 500 do
		motor.left.target = 300
		motor.right.target = 300
	end
	when prox.horizontal[2] > 2000 do
		if state[1] == 1 then
			motor.left.target = 0
			motor.right.target = 0
		end
	end
	when prox.horizontal[0] < 1000 and prox.horizontal[1] < 1000 and prox.horizontal[2] < 1000 and prox.horizontal[3] < 1000 and prox.horizontal[4] < 1000 do
		motor.left.target = 250
		motor.right.target = 250
	end

stephanemagnenat, 2015-07-15:
hum, I think I have done an over zealous optimisation

stephanemagnenat, 2015-07-15:
Corrected in commit cd28052, sorry for inconvenience.

stephanemagnenat, 2015-07-15:
Please test and close.

riedo, 2015-08-10:
All OK imo

closed 2015-08-10 by riedo