aseba-community/aseba #396: Possibility to comment lines in VPL?

riedo, 2015-08-10:
Just an idea: could it be interesting to be able to comment lines in VPL? or store them somewhere for later use?

stephanemagnenat, 2015-08-10:
This is an interesting idea. I think it should be considered in the context of sharing the code, maybe users could have their personal "scratch notebook" in which they can store (and annotate) their personal programs or the code they use often?

FrancescoMondada, 2015-08-10:
Yes, very good idea.

------------------------------
Prof. Francesco Mondada
Mobile robots architecture & design

Ecole Polytechnique Fédérale de Lausanne (EPFL)
Laboratoire de Systèmes Robotiques (LSRO)

Postal address:
Francesco Mondada 
EPFL  - STI  - IMT - LSRO 
bureau MEB3426, Station 9
CH-1015 Lausanne, Switzerland

tel: +41 21 693 73 57
fax: +41 21 693 78 07
http://lsro.epfl.ch

stephanemagnenat, 2015-08-10:
I think that there is a very important transversal skill we can teach to children using such a feature: information organisation. I am always amazed that even university students are not able to give proper variable names! So with such a "scratch notebook", we can teach the following transversal concepts:
- modularising and re-using one's work
- documenting one's work
- proper referencing of re-used other's work – intellectual property education

motib, 2015-08-10:
One can get inspiration from Scratch 2.0. It has a "backpack" - an area on
the screen to which you can drag and drop blocks, scripts, costumes,
sprites. These are stored in the cloud and thus available between projects
and well as for sharing.

stephanemagnenat, 2015-08-10:
I agree, Scratch is good to keep track of sharing!

riedo, 2017-11-02:
I was about to open this issue again as someone yesterday asked for this feature :-)
They said they wanted to deactivate a few lines do debug their program and test only parts of it without losing the whole code.

stephanemagnenat, 2017-11-02:
It is an interesting feature indeed, could you please open it in VPL2 https://github.com/aseba-community/thymio-vpl2/issues ?