aseba-community/aseba #399: disconection on windows

mbonani, 2015-08-12:
Actually on windows when you disconnect there is no reaction until you try to communicate with Thymio (for instance if auto refresh is there no problem) .
This is more a problem on VPL...
Could we imagine to have a ping every 1s to control that robot is always there, a small command that make very small data?

stephanemagnenat, 2015-08-13:
> Actually on windows when you disconnect there is no reaction until you
> try to communicate with Thymio (for instance if auto refresh is there no
> problem) .

Indeed it is a problem on Windows, as the Windows serial port API does
not provide a way to detect disconnection if nothing is written.

> This is more a problem on VPL...
> Could we imagine to have a ping every 1s to control that robot is always
> there, a small command that make very small data?

This is a good idea!

I see two solutions to implement this:

1. Send an empty message of null data length with invalid type (0xFFFF).
This one should be dropped at the level of the checking of the message
size in `AsebaProcessIncomingEvents()` in `vm-buffer.c:230`. The total
message size would be 6 bytes.

2. Send an empty message of invalid type (type 0xFFFF) to an invalid
destination (0xFFFF). This one should be dropped at the level of the
checking of the destination id in `AsebaVMDebugMessage()` in `vm.c:720`.
The total message size would be 8 bytes.

I do not have the source code of Thmyio's firmware with me, nor do I
have internet access, so I cannot check which of these would work with
the firmware. They would both work when using `vm-buffer`. If both would
work with the firmware, I would vote for solution 1. as it forwards the
dummy message less deep into the VM than 2.

Any comment (Philippe?) is welcome!

mbonani, 2015-08-18:
a empty message is perhaps not the optimal solution when we think about the wireless. How should we manage that robot are still in the network... or entering it... I do not know well aseba messages but having that in mind could guide the choice

stephanemagnenat, 2015-11-24:
This is fixed by commit 220dfe9676586a614dbaa26346eeabd1aed949f9.

closed 2015-11-24 by stephanemagnenat