aseba-community/aseba #400: Complete asebahttp integration

davidjsherman, 2015-08-15:
Add menu entry for "Web Tools for Thymio" that runs asebahttp with the thymio_motion.aesl file parameter, the default TCP port 3000 (for the HTTP interface), and ser:name=Thymio-II; and asks user's web browser to open the javascript welcome page.

Depends on the following issues in https://github.com/davidjsherman/aseba/milestones/Version%201.4:
- davidjsherman/aseba#8
- davidjsherman/aseba#7
- davidjsherman/aseba#6

davidjsherman, 2015-08-16:
After thinking about it, I believe that it is reasonable that both the web interface (davidjsherman/aseba#8) and Aseba Studio can communicate with the robot at the same time. If *asebahttp* is run as described above, it will lock the USB connection and prevent Aseba Studio from communicating with the robot.

Consequently, I think that a better solution would be that the menu entry run *asebamedulla* in addition to *asebahttp* as described above. In that way Aseba Studio can share the connection to the robot using dashed target tcp:;port=33333, and *asebahttp* will relay Aseba tcp:;port=33333 to HTTP localhost:3000.

stephanemagnenat, 2015-08-16:
I agree but see two open questions:
* Should one run /asebamedulla/ or /asebaswitch/? To former provides a DBus bridge but is heavier and seems to have some bugs while the later is simpler. Also, /asebamedulla/ does not run on Windows.
* If we answer switchbto the previous question, shouldn't the switch feature be provided by /asebahttp/ itself?

riedo, 2015-08-18:
asebamedulla is currently not packaged for Mac OS either. Why not use asebaswitch?

stephanemagnenat, 2016-01-26:
Asebahttp is now working with protocol 5 (see 31d63747853e5ed9c5c3e215102730c887d7afec), so I retarget this issue for Aseba 1.6.

davidjsherman, 2018-01-11:
Asebahttp acts as an asebaswitch since c682c2103d6e10d4c65903f541703af2abfdbf55.

stephanemagnenat, 2018-02-02:
Do we still want this in addition to the launcher (#728)?

cor3ntin, 2018-02-02:
I'd say put all the issues vaguely related to the launcher or the switch in the launcher milestone and we will cleanup from there as we refine the use cases/ ux of the launcher ( and no, I don't think we want that, even though it's actually done on osx/win32, it will need to be improved, a lot )