aseba-community/aseba #402: Implement Thymio2 in target Enki playground

davidjsherman, 2015-08-15:
Currently the Thymio2 robot in Enki is an alias for the Epuck. For automatic testing it would be valuable to have a Thymio-II emulator in the Enki playground, that we could use both for validation and for long term testing.

In order to stress test movement and odometry [thymio_motion.aesl](https://github.com/davidjsherman/inirobot-scratch-thymioII/blob/master/thymio_motion.aesl) and asebahttp/asebascratch we need at **minimum**:
- a motor event at 100 Hz, for the movement queue;
- native functions math.muldiv, math.cos, math.dot, math.min, math.clamp;
- sensors prox.horizontal and prox.ground; and
- motor variables motor.right.target and motor.left.target (which needed for the playground but not for odometry since they are read and written like ordinary variables)

To completely test [thymio_motion.aesl](https://github.com/davidjsherman/inirobot-scratch-thymioII/blob/master/thymio_motion.aesl) as it is used for Scratch, we would need **optionally** the native functions for LEDs and sound
- leds.bottom.right, leds.bottom.right, leds.top
- sound.system, sound.freq

We don't really need the rest; the events and reporter variables for the other LEDs, sound recording, rc.tx/rc.rx etc will just not work.

A simple playground with a large flat table, black borders, and perhaps two straight black lines on side to test the calibration code would be useful for automatic testing.

stephanemagnenat, 2016-05-11:
Basic support is implemented now in master. See issues tagged "Simulator" for full support.

stephanemagnenat, 2016-11-17:
Minimum support is available as from 2e0af266227a66d02b4dfc22e0006e5f6a648bbb.