aseba-community/aseba #407: Reset subroutine

motib, 2015-08-23:
I wish that Studio had "call reset" that would do what VPL automatically does:
# reset outputs
call sound.system(-1)
call leds.top(0,0,0)
call leds.bottom.left(0,0,0)
call leds.bottom.right(0,0,0)
call leds.circle(0,0,0,0,0,0,0,0)
This would make programs much more concise.

stephanemagnenat, 2015-08-26:
Do you mean a button like the stop button, or rather a subroutine? If so, I see that wish as a dependency of issue #278 .

motib, 2015-08-26:
Hi Stephane,
I mean a builtin subroutine just like "call leds.top()". The reason is that I often write such a routine to reset the Thymio between stages of an algorithm, but then I can't call it also in the initialization :-(, so it leads to duplicate code. I don't see this as related to 278, because that is a wish for a general feature - that may not be feasible to implement - while is a request for a specific function that I imagine shouldn't be hard to implement.
Moti

stephanemagnenat, 2015-08-26:
I understand, so this concerns the firmware. I have added a corresponding issue, see aseba-community/aseba-target-thymio2#10

mbonani, 2015-11-16:
A such function was implemented and call system.softreset() and is link to https://github.com/aseba-community/aseba-target-thymio2/issues/10.
But we also make clean reset of all robot at each load or reset : https://github.com/aseba-community/aseba-target-thymio2/issues/8

We are not sure now that the softreset is useful? 
Please comment

motib, 2015-11-16:
I encountered several cases in Scratch (which is also event-driven) where the lack of consistent initialization was a problem. The fact that VPL generates this code indicates that this was considered important here too and AESL users should have the same capability. Initialization is a basic concept that is not easy to teach and learn, so any simplification is helpful.

Thanks for implementing this!!

mbonani, 2015-11-16:
To clarify my question, we have now:
- at each reset or load of program, an automatic reset that really clear every thing, form variable to state of the robot. User has nothing to do. At each load the Thymio is as first connection. 
- a system.softreset() function that clear all leds, stop the Ir.communication and activate the basic behavior (the led near the proximity and the sound when press à button. 

The second was one of your wishes, but at same time we develop the first solution. We are now not sure if both are needet. The second is only usefull if in the middle of your program you want to softreset your robot.... Is that the case? if not we will not provide second one to users.

motib, 2015-11-16:
I find it useful to define one button as a reset button, so that a program can be run repeatedly without going back to the computer to load and run. This was inconvenient because the initialization code had to be repeated both in the initialization code of the program and in the event handler for the button. If all I have to do is write "system.softreset()" then the problem disappears.

One more question: at one point I think I asked about the initialization of the IR communications. Does the new reset include that??

mbonani, 2015-11-17:
hello,
yes it turn off the ir communication, do you want that it also clear all ir.comm values?
regards

motib, 2015-11-17:
Yes. I think that I should be able to touch a button and reset everything in the event handler with one call.
Thanks!

mbonani, 2015-11-24:
hello,
we still have some doubt:
Do you want that the "reset function" reset every thing?  Even the the program counter so that we really begin as from a load, or you prefer a more soft one like describe here? 

Also now when you turn off ir.comm it also clear prox.comm.rx and prox.comm.tx independently of the reset. 
Regards

motib, 2015-11-24:
Hi Michael,
I never thought about resetting the PC... Although it sounds tempting, I'm not sure that it is a good idea. The FW should reset the HW, while the PC is under control of the SW. I hope someone else can help discuss this issue before you decide.
Moti

stephanemagnenat, 2015-11-24:
To summarize the situation, now when a program is loaded or the "reset" button is clicked in Studio, the robot's HW state is reseted to its status when the VM was first started. In that situation, I am not sure it is useful to have an additional reset native function, as this behaviour can easily be implemented in a subroutine. What do you think?

motib, 2015-11-24:
I suggested that reset be available within the program, so that the user _doesn't_ have to click anything in Studio, but can simply touch a button or tap the Thymio. It is true that the behavior can be implemented in an reset subroutine, provided that the subroutine can be called initially and not just from an event handler. I found it frustrating to have to duplicate lots of code, which makes the code longer and less structured than it should be. I am agnostic as to which solution is chosen: enabling subroutine calls initially or encapsulating reset in a single native call.

retornaz, 2015-11-24:
Well, if the VM and the robot state is reseted every time you upload new bytecode then I don't see any usecase to have to call this subroutine in the init.

But If we provide a way to soft-call a full VM reset we have to correcly handle the case were the user would call this function in the initialisation as we could end up in a live lock.

Anyway, the point is now clear for me that this has to be handled at a generic level by the VM, if done. Thus a custom half-reset in the thymio-specific native function is not the solution.

mbonani, 2015-11-24:
I will close this issue, letting people use next firmware. If afeterwall it is still needet, we can implement the VM reset.

closed 2015-11-24 by mbonani