aseba-community/aseba #424: Block in VPL to control Thymio with a remote controller

cbarraud, 2015-09-15:
As a new step toward diversification of usage of Thymio for its users, I think that a block allowing some controls with a remote controller (maybe only the official Thymio one) would be a great idea. 

[image]

The Thymio remote controller from mobsya has a GO and a STOP button, four arrows, 10 digits and also a + and a - button. We could maybe have just the GO/STOP and arrows. In advanced mode another block could allow access to digits and + / - buttons. If we keep the same channel and configuration of the remote controller, it could be implemented (and probably quite easily... no?)

What do you think about this?

Christophe

FrancescoMondada, 2015-09-15:
If this is not limited to this remote, I think it's a great idea.

					Francesco

------------------------------
Prof. Francesco Mondada
Mobile robots architecture & design

Ecole Polytechnique Fédérale de Lausanne (EPFL)
Laboratoire de Systèmes Robotiques (LSRO)

Postal address:
Francesco Mondada 
EPFL  - STI  - IMT - LSRO 
bureau MEB3426, Station 9
CH-1015 Lausanne, Switzerland

tel: +41 21 693 73 57
fax: +41 21 693 78 07
http://lsro.epfl.ch

cbarraud, 2015-09-15:
I just proposed to limit to simplify the code. The remote usually have different codes and buttons... could it be adaptable? Like in advanced mode we could match buttons, like press a button, then the robot get the command and address and adapt the block... (seems complicated)

We could begin with the thymio remote and maybe one or two classical others...

mbonani, 2015-09-15:
No to complicated
It is easier to do as the purple mode does that  accept different command
for "same" button. We limit the buttons of thymio remote but accept all
command code that are generated by other remote for this same buttons.
There are not so much.

stephanemagnenat, 2015-09-15:
The design rational of VPL calls for a limited number of blocks. The motivation stems both from the research in visual design and in education: a limited amount of information must be presented to the beginners. Therefore, we have to be careful not adding too much blocks to VPL. The next ones we want to add are the ones for infrared-based communication as this is a strong feature of Thymio that is vastly underexploited to date.

It is not clear to me what the remote-control brings from a programming perspective. The whole story about the approach in our product is to shy away from remote control and program reactive and event-based behaviours in the robot. So this in light, I do not see adding a block for the remote control as a priority.

motib, 2015-09-16:
My young grandchildren have a lot of fun playing with RC toy cars and driving vehicles on their PlayStation, but I don't see any educational purpose to these activities. VPL already supports three human-originated events (buttons, tap, clap) so RC control doesn't seem to add any conceptually novel functionality.

I prefer Stephane's idea of adding communication because it is a novel functionality that is currently not supported in VPL.

cbarraud, 2015-09-16:
That's exactly the point. Having fun ;)

In my opinion, everything should not be thought only in an educational way. Fun is also important. If you have fun with something, it motivates you a lot to go deeper in the less fun aspects. I strongly believe that Thymio will have a good success only if we find a balance between education and fun.

This block won't be very useful in terms of programming skills, but its goal is not to be useful. Its goal is to allow a richer use of Thymio. It will allow straightforward easy new things to code, like an RC toy car. Sure, it's not as deep as the IR communication one but I really believe that Thymio now needs more a block allowing to boost the play and fun access instead of a block allowing complex control with IR communication.

If you want to use an remote controller now, either you use the purple mode (which is not easy at all to control), or you code it in the hard way. I strongly believe that this block is mandatory, especially now that we want to diversify the use of Thymio and to extend the "use time" of the robot.

What is your opinion ?

motib, 2015-09-16:
> That's exactly the point. Having fun ;)

Agreed, but the question is "what kind of fun?" Seymour Papert defined "hard fun" (http://www.papert.org/articles/HardFun.html) based on a comment by a _first-grade_ kid writing Logo. In my opinion, the current pre-programmed behaviors of the Thymio are sufficient for "soft fun" and kids can jump quickly into VPL for "hard fun."

For the opposite approach, see the new Makewonder robots Dot and Dash. They have three apps (Go, Path, Xylo) that provide "soft fun" far beyond what the Thymio can currently do. However, their programming app (Blockly) is extremely limited. In VPL, two lines are enough to implement many Braitenberg creatures (which can't be done with the Dash) and I think that the problem-solving required will be much more fun than controlling a robot with an RC. Kids usually tire of such simple toys after a couple of hours...

In the end, this is an empirical question that should be looked into before making a decision.

cbarraud, 2015-11-25:
I am coming back about this Remote controller VPL block :)

In a user experience perspective, playing with the pre-programmed behaviours is quite fun, yes. Playing with the remote controller with the purple pre-programmed behaviours brings another dimension, ok. But programming your robot so that it does the same as your RC car is priceless. Humans always compare things, the 100% pure creativity does not exist in my opinion. Creativity for me is more like taking parts of things you saw, heard, touched, etc, and bringing them together in an original way. For this reason, being able to recreate something everybody know (like an RC car), and being able to say: "I programmed it", that's great! 

I really believe that this block is missing. People always ask us: "What can we do with this remote controller?" and then we have to say: "Well, just learn how to code it with text, it's easy, there are juste two variables to test.......". In one word, it's way underused, and I think that's really a bad thing for Thymio.

I totally agree that this block is not fundamental in a programming perspective, but I really believe it would be a great improvement in terms of user experience.

(and for the programming perspective, I used a lot of remote controller in the Thool project, and it's a real asset when it comes to programming different behaviours and being able to choose from one another)

FrancescoMondada, 2015-11-25:
I do not know why you need to make a great theory about "what is 100% creativity" etc and why the remote control has nothing to do with programming. The RC allow to have distant buttons, period. If local buttons are good for programming, distant buttons should also be good and would add the distance activation as a plus. On the other hand I agree that we have to see which priority we should give to this, as the programming people are totally overloaded. Then Christophe you should perhaps look at this issue…

									Francesco

------------------------------
Prof. Francesco Mondada
Mobile robots architecture & design

Ecole Polytechnique Fédérale de Lausanne (EPFL)
Laboratoire de Systèmes Robotiques (LSRO)

Postal address:
Francesco Mondada 
EPFL  - STI  - IMT - LSRO 
bureau MEB3426, Station 9
CH-1015 Lausanne, Switzerland

tel: +41 21 693 73 57
fax: +41 21 693 78 07
http://lsro.epfl.ch

yduponchel, 2015-12-27:
Adding to the discussion about the remote control, my daughter (6 years old) just got her Thymio for Xmas, she liked everything a lot, but the first thing she asked was: "can I program the remote control?".  

As far as "education" is concerned, I believe it would be a great addition because it is in fact the only way between fully programmatic (VPL only) and fully manual (purple mode, with remote control), and when you want to teach young kids, it allows you to build complex logic, but split it in many small blocks that they can understand and then sequenced manually.

Bottom line, I fully support Christoph's initial proposal (with simple and advanced mode), and I am convinced it would help adoption with the younger ones without adding complexity (the arrows/stop are already present, but only available on the device itself - it even surprised me that the remote was not duplicating these buttons)

cbarraud, 2016-04-26:
Following a discussion we had here, we could maybe find a compromise ;)

The idea is not to add a new block, but to add a feature an existing one, the buttons block.

An example. This VPL blocks pair:

[image]

currently generate this code:

`onevent buttons
	when button.forward == 1 do
		motor.left.target = 500
		motor.right.target = 500
		emit pair_run 0
	end`

instead, we could make it generate this kind of code:

`onevent buttons
	when button.forward == 1 do
		motor.left.target = 500
		motor.right.target = 500
		emit pair_run 0
	end

onevent rc5
	when rc5.command == 2 or rc5.command == 80 or rc5.command == 32 do  # Same as for the purple mode
		motor.left.target = 500
		motor.right.target = 500
		emit pair_run 0		
	end`

So if you press the forward button or the forward arrow of the pad of the remote controller, it would create the same reaction.

This
[image]

Would be equal to the buttons on Thymio.

What do think about that?

stephanemagnenat, 2016-04-28:
This idea has the advantage of adding a feature without adding a new block, which is quite nice. But it has the drawback of making the generated code less readable, which is unfortunate because it is already not so easy to read (for instance for the sound block). Moreover, it splits the generated code of a block at two different places, which breaks the current scheme of selecting a pair and viewing the generated code. So I will definitely consider it for VPL2, but for the current one, I am not convinced.

cbarraud, 2016-12-21:
As it has been implemented, I close the issue :)

closed 2016-12-21 by cbarraud