aseba-community/aseba #426: Ubuntu packages have too large install footprint

stephanemagnenat, 2015-09-17:
Currently, the PPA built of Aseba have an install footprint of more than 300MB, this is too much!

Things can be improved by:
* using shared libraries for Aseba
* not installing example binaries and experimental programs such as:
 * aseba-qt-dbus
 * aseba-qt-gui
 * aseba-test-natives-count
 * asebabotspeak
 * asebatest

stephanemagnenat, 2016-01-25:
Switching to dynamic libraries is too much work for the short 1.5 deadline, pushing to 1.6

cor3ntin, 2018-02-06:
the following libraries are good candidates for shared libraries

```
3,3M    aseba/clients/studio/libasebaqtcommon.a
3,1M    aseba/clients/studio/plugins/libasebaqtplugins.a
2,0M    aseba/common/libasebacommon.a
1,3M    aseba/common/zeroconf/libasebazeroconf.a
2,5M    aseba/compiler/libasebacompiler.a
```

davidjsherman, 2018-02-06:
But not `vm` and `vmbuffer`, *cf.* https://github.com/aseba-community/aseba/issues/795#issue-291022563?

cor3ntin, 2018-02-06:
@davidjsherman meh, they are extremely small libraries, a few K, no real need for them to not be static.

davidjsherman, 2018-02-06:
Please document the use cases where more than one or two processes will be using the shared libraries that you have identified.