aseba-community/aseba #427: IR communications disabling and initialization

motib, 2015-09-23:
1. Clicking "Stop Thymio" should disable communications.

2. There is no way to clear prox.comm.rx._intensities[] and hence to clear prox.comm.rx. They are only cleared by turning the Thymio off. This is important so that a program can reset itself or to have different behaviors in different parts of the program.

stephanemagnenat, 2015-09-23:
Point 1. can be addressed easily, and I agree about it.
Point 2. is a firmware issue, and should be reported here: https://github.com/aseba-community/aseba-target-thymio2/issues/new

motib, 2015-09-24:
1. Thanks.
2. Done.

stephanemagnenat, 2015-11-24:
We just looked again into 1. Disabling RF communication actually involves calling a native function, so it is not trivial to do after all (one must load a small code doing so). Do we really need it, now that re-loading the code or clicking on reset in Studio disables the IR communication network as well?

motib, 2015-11-24:
I don't see any reason to insist on this, so I'm OK if you want to close the issue.

closed 2015-11-24 by stephanemagnenat

retornaz, 2015-11-24:
Well, no, you don't have to upload a small bytecode if the firmware-reset is called from the VM reset (pull request pending, as it needs some cleanup, but should be merged soon I hope).
You will just have to send the VM reset message.