aseba-community/aseba #435: Thymion dont reconnect

Thomasmilowiz, 2015-10-21:
I have done a Programm with the thymion. I compiled it with an USB-cable. But if I put off the cable from thymion and put it again in, the program doesnt find thymion anymore. But i hear the usb-connecting/disconnecting-sound from Windows everytime. i am using win 10 and asiba 1.4.

FrancescoMondada, 2015-10-26:
Got similar feedback from other users, for instance here from a training in Italy:
On a few desktop there were recurrent problem of VPL hanging with a white screen after the compilation and test of a program. As soon as Thymio was detached from USB VPL was not able to reconnect. 
We had a procedure:
a) stop Thymio
b) reconnect the USB cable
c)close and restart VPL and reconnect Thymio.
Equipment used: Desktop with Windows XP, Aseba 1.4 nightly build of this weekend, Thymio some with firmware 8 and some 9.

stephanemagnenat, 2015-10-26:
Could it be defect cables, and weird reactions from the operating system?

Thomasmilowiz, 2015-10-26:
No... we got new tymions and i tried it with a few thymions and it was 
always the same....

Gesendet mit AquaMail für Android
http://www.aqua-mail.com

mbonani, 2015-10-27:
I just test now. It is a problem of windows 10. Disconnections is not  detect.

mbonani, 2015-10-27:
@FrancescoMondada 
This case seems to be different. It is reletad to issue #399  
In Windows the disconnections is not detect until there is a data transfer. In VPL it is only the case when you click on play. At this moment you have the reconnection message.
So in actual VPL and Windows system, when you disconnet, before reconnect press the play button to make apear the reconnection message. Also if you have the reconnection window apear and you robot is already connect, diconnect and reconnect it should work.

mbonani, 2015-10-27:
usbser driver was update in win 10. This should explain why disconnection do not react has previous version. More info here:
https://msdn.microsoft.com/en-us/library/windows/hardware/dn707976(v=vs.85).aspx

dougbergman, 2016-10-20:
Great to hear others have experienced the same thing. We love these robots and our students love them. If we could fix this, rather than have to close program and reconnect We are using Aseba Studio, not VPL(for text-based coding). Any progress on this issue?

mbonani, 2016-10-20:
Sorry for bad news but no progress at this time. We need a Win 10 specialist to handle this. The only fix we have is to use Wireless Thymio so that you do not need reconnection or that you close save and clse the program before ( I know not user friendly at all)
But we see the needs and will try to find a solution. Open to suggestions.

dougbergman, 2016-10-25:
hey....thanks for response. So we have 30 units, but do not think they are
wifi version is there a way to make these units work with dongle?

mbonani, 2016-12-05:
Hi,
Normally all Thymio could be update to Wireless version. You can bought an upgrade kit like this one:
http://www.mobsya.org/en/kit-wireless-diy.html
Regards

swithers19, 2017-03-12:
Hi there,

So it has been two months and there has been no developments with a solution. This is making it really difficult to use the robots as it is a nightmare for kids to disconnect, connect, ect. Is there any movement on a solution or 'hack' solution.

Cheers

stephanemagnenat, 2017-04-28:
I have opened an issue on Dashel regarding this problem (/aseba-community/dashel/issues/22). As a summary, the problem is that the built-in Windows 10 `rfcomm` driver does not generate write (or read) errors on the associated stream when the physical USB-serial device is disconnected. This is a serious flaw in the driver.

ypiguet-epfl, 2017-06-15:
Doesn't seem to be limited to the serial port. Try with asebaswitch between Aseba Studio and the Thymio as follows:
- connect the Thymio to the PC
- start `cmd.exe`
- cd to the directory containing your Aseba applications (typically `C:\Program Files (x86)\AsebaStudio`)
- start asebaswitch with something like `asebaswitch.exe "ser:device=COM6"` (you can check which COM port to use with Aseba Studio first)
- start Aseba Studio
- connect to TCP localhost 33333; you should see a window with Thymio's built-in variables and the code editor
- kill asebaswitch with ctrl-c in cmd.exe; in Aseba Studio, the window should become empty without any visible attempt to reconnect
- start asebaswitch again; in Aseba Studio, nothing happens.

On Linux, restarting asebaswitch causes Aseba Studio to reconnect.

I tested that with Windows 10. Does it work on Win7?

stephanemagnenat, 2017-06-15:
If this test case is confirmed, that would mean that the whole disconnection notification (including for TCP) for the native API is broken in Windows 10.

stephanemagnenat, 2017-06-15:
@swithers19 we are working on a work around for the serial port. The problem is that the operating system does not report disconnection when they happen as it should. It is a bug in the operating system.

mbonani, 2017-10-10:
I stress test reconnection with the "econnect" example of dashel. It works fine, there is I think now a problem in studio.

stephanemagnenat, 2017-10-10:
Ok, then it might be a threading issue. Let's talk live about this then.

mbonani, 2017-12-15:
this fix now

closed 2017-12-15 by mbonani