aseba-community/aseba #441: Catching Thymio events in Python through DBus

pierreboudes, 2015-10-26:
The code exemple of the Python bindings for D-Bus examples/clients/python-dbus/aseba.py introduced in #290 to resolve  #286 is quite confusing. 

How is it supposed to work after entering the infinite loop `gobjects.MainLoop()` (`loop.run()` is called when `__enter__`'ing the with statement) ? 

If we postpone the `loop.run()`, the code correctly fetch the values (`"acc"` and `"temperature"`) but trying to capture `"timer0"` fails with a `no event named timer0` error. With a numerical id instead of an event name, it is executed without throwing any error but the callback is never called.

This is the only piece of documentation on events handling and after tweaking a bit with it I am still unable to capture an event emitted by Thymio. Is it possible to have some hints on how to program a simple example, let say, of a Python code which react to a `button.backward` emitted by a Thymio ?

I c

stephanemagnenat, 2015-10-26:
@severin-lemaignan ?

pierreboudes, 2015-10-26:
Thanks, for your quick reaction. I look forward for an enlightening answer by @severin-lemaignan about that loopy thing.

I have finally figured out how events are supposed to work their way through DBus to the Thymio and back. A straightforward and tiny example, without error management code, is here : https://github.com/pierreboudes/pyThymio/blob/master/watch_back.py . 

I think that the documentation should mention that one need to load into the Thymio a suitable aesl code forwarding any interesting local event. Before trying to capture events like 'timer0' it looks like one needs to give them global names.  I will work a bit on writing improved example codes, feel free to copy and modify it at any time and to correct my false asumptions.

severin-lemaignan, 2015-10-26:
Hello! Oulala... I wrote this code quite a long time ago, I did not actively use Aseba since several years, and I do not remember the details...
The Python bindings are not great because of their reliance on DBus -- which is painful to use.
The best thing I can do (except recommending to use something else than Aseba ;-) naaaah I'm being mean! ;-) ) is to point you to the code I wrote to control the Ranger robot from Python. Here is the relevant part: https://github.com/chili-epfl/pyranger/blob/master/src/ranger/lowlevel/ranger_aseba.py#L115

I hope it helps!

pierreboudes, 2015-10-26:
Ok, thanks !  I will look at it.