aseba-community/aseba #447: Screenshots can be corrupted by reordering VPL code

ddoak, 2015-11-02:
See these example screenshots:

[image]
and
[image]

To reproduce:
Create a VPL program with instruction lines of different lengths (e.g. 1 vs 2 action blocks). Save a screenshot of the program, then reorder the instruction lines by dragging one on top of the other and then screenshot again. The second screenshot will contain un-refreshed areas where the graphics did not overlap.

Only tested so far on Mac (OS X 10.9.5) (100% reproducible on two different machines)

stephanemagnenat, 2015-11-02:
That is fascinating. Does it happens with both SVG and PNG?

ddoak, 2015-11-02:
Just checked - SVG seems to be fine, PNG is broken.

(I noticed it earlier today when I was trying to make screenshots for #446 - I was dragging the blocks to reorder the code)

I also managed to do this earlier (png) by resizing the VPL window - which looks like a screen buffer issue -  but I couldn't find simple clear steps to reproducing it.

[image]

stephanemagnenat, 2015-11-02:
Are you using Windows? On Linux I could not reproduce this bug.

ddoak, 2015-11-02:
Mac (OS X 10.9.5)  (100% reproducible on two different machines)

I can't reproduce it on Win XP.

stephanemagnenat, 2015-11-02:
It looks like a Qt bug to me.

stephanemagnenat, 2016-12-21:
Is this bug still present with Qt 4.8.7 as used in the Windows 1.5.5 packages?