aseba-community/aseba #451: Crash when display plot

cbarraud, 2015-11-09:
Hi,

I tested the plot feature of Studio with the built Windows: Git 48fedc4 of the 5th of November and it crashes everytime I click on plot.

Here is the code I put into Thymio:

*onevent prox
emit acc_measure acc[0]*

With an global event called *acc_measure* with 1 variable. 

It just writes: "Aseba stopped working" and then, I'm forced to quit.

I installed several versions of Aseba on my laptop in parallel (1.3, 1.4, etc), I don't know if it could be that...

++

stephanemagnenat, 2015-11-09:
It seems nightly builds of windows are unstable, could you try using Linux or OS X? There was no change on this part of code for years.

cbarraud, 2015-11-09:
I guess that's it. I will try to find a tester on Linux or OS X.

++

stephanemagnenat, 2015-11-24:
This is fixed by using the proper version of `libqwt` (same Qt version, same debug/release flag).

closed 2015-11-24 by stephanemagnenat