aseba-community/aseba #456: Blockl should be translated

stephanemagnenat, 2015-11-24:
Missing:
* French
* Italian
* Spanish

frankyb88, 2015-11-27:
#0  0x00007ffff6a3212d in poll () at ../sysdeps/unix/syscall-template.S:81
#1  0x00007ffff5e05fe4 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff5e060ec in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff75df7a1 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#4  0x00007ffff75b10af in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#5  0x00007ffff75b13a5 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#6  0x00007ffff75b6b79 in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#7  0x0000000000438f4d in main (argc=4, argv=0x7fffffffde58) at /var/lib/jenkins/jobs/aseba.ubuntu-trusty-amd64/workspace/source/switches/medulla/medulla.cpp:724

frankyb88, 2015-11-27:
something like this ? (I am not familiar with gdb)

stephanemagnenat, 2015-11-27:
Yes, but this is posted in the wrong issue...

stephanemagnenat, 2016-02-10:
Implemented in commit fc0b32e1e4ee07185123972cb7a35a2add34458b

closed 2016-02-10 by stephanemagnenat