aseba-community/aseba #457: asebamedulla with protocole 5, problem with mode dump (-d) using bluetooth connection

frankyb88, 2015-11-27:
asebamedulla crashes everytime when using a bluetooth link with the new protocole (5) and when using the dump (-d) option:

[Fri Nov 27 14:42:59 2015 922] Incoming connection from ser:device=/dev/rfcomm5;baud=115200;stop=1;parity=none;fc=none;bits=8;dtr=true
[Fri Nov 27 14:43:00 2015 122] 0xa000 presence from 0 : protocol version 5

asebamedulla can connect only without the dump option.

stephanemagnenat, 2015-11-27:
Are you using latest Aseba master? (protocol version 5)

frankyb88, 2015-11-27:
yes

stephanemagnenat, 2015-11-27:
Can you provide a backtrace using a debug version of aseba please? Thanks!

frankyb88, 2015-11-27:
#0 0x00007ffff6a3212d in poll () at ../sysdeps/unix/syscall-template.S:81
#1 0x00007ffff5e05fe4 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2 0x00007ffff5e060ec in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3 0x00007ffff75df7a1 in QEventDispatcherGlib::processEvents(QFlagsQEventLoop::ProcessEventsFlag) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#4 0x00007ffff75b10af in QEventLoop::processEvents(QFlagsQEventLoop::ProcessEventsFlag) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#5 0x00007ffff75b13a5 in QEventLoop::exec(QFlagsQEventLoop::ProcessEventsFlag) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#6 0x00007ffff75b6b79 in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#7 0x0000000000438f4d in main (argc=4, argv=0x7fffffffde58) at /var/lib/jenkins/jobs/aseba.ubuntu-trusty-amd64/workspace/source/switches/medulla/medulla.cpp:724

frankyb88, 2015-11-27:
sorry for that...

stephanemagnenat, 2015-11-27:
so it seems it is the wrong thread, this one is blocked in `poll()`, I do not think it is the one which segfaulted. It is a segfault, right?

frankyb88, 2015-11-27:
NO it is not a segault, sorry for the use of "crash", it is just pending in this condition but the device is not connected, and the node description does not display on the terminal

stephanemagnenat, 2015-11-27:
Ah! Yes, that's true, asebamedulla was not updated for protocol 5. I'm on it...

frankyb88, 2015-11-27:
cool thank you!

stephanemagnenat, 2015-11-27:
Minimal support added in commit 84605c9, no dynamic connection/disconnection yet.

stephanemagnenat, 2016-01-21:
The latest version (commit 6be7f25cc0e40339c2f9b620d0bfb328118368bb) should support connection/disconnection of targets within an Aseba network. Automatic reconnection to Dashel targets is implemented in asebahttp2, into which medulla will be merged in the future. But that might not happen within the 1.5 release cycle.

Could you please test the current master version and, if it solves your problem, update/close this bug? Thank you!

stephanemagnenat, 2016-01-25:
Any update on this testing?

frankyb88, 2016-01-25:
I tested it, from what I see it works as before. All the functionalities are the same and work, but sometimes when connecting to several nodes I still have some issues that look the same as before (except that now it seems to several time to reconnect, but sometimes fail anyway). For example:

fishbot@fishbot-Precision-WorkStation-T3500:~/aseba/build/switches/medulla$ ./asebamedulla -v -d "ser:device=/dev/rfcomm0" "ser:device=/dev/rfcomm1"
[Mon Jan 25 12:12:23 2016 718] Incoming connection from ser:device=/dev/rfcomm0;baud=115200;stop=1;parity=none;fc=none;bits=8
[Mon Jan 25 12:12:23 2016 719] Incoming connection from ser:device=/dev/rfcomm1;baud=115200;stop=1;parity=none;fc=none;bits=8
[Mon Jan 25 12:12:24 2016 718] 0xa011 list nodes from 0 : protocol version 5
error while writing message
error while writing message
[Mon Jan 25 12:12:24 2016 718] 0xa000 presence from 0 : protocol version 5
error while writing message
error while writing message
[Mon Jan 25 12:12:25 2016 718] 0xa011 list nodes from 0 : protocol version 5
error while writing message
error while writing message
[Mon Jan 25 12:12:25 2016 718] 0xa000 presence from 0 : protocol version 5
error while writing message
error while writing message
[Mon Jan 25 12:12:26 2016 718] 0xa011 list nodes from 0 : protocol version 5
error while writing message
error while writing message
[Mon Jan 25 12:12:26 2016 719] 0xa000 presence from 0 : protocol version 5
error while writing message
error while writing message
[Mon Jan 25 12:12:27 2016 719] 0xa011 list nodes from 0 : protocol version 5
error while writing message
error while writing message
[Mon Jan 25 12:12:27 2016 719] 0xa000 presence from 0 : protocol version 5
error while writing message
error while writing message
[Mon Jan 25 12:12:28 2016 719] 0xa011 list nodes from 0 : protocol version 5
error while writing message
[Mon Jan 25 12:12:28 2016 719] 0xa000 presence from 0 : protocol version 5
error while writing message
[Mon Jan 25 12:12:28 2016 804] Abnormal connection closed to ser:device=/dev/rfcomm0;baud=115200;stop=1;parity=none;fc=none;bits=8 : File write I/O error. Transport endpoint is not connected
[Mon Jan 25 12:12:28 2016 804] 0x900c node present from 30 : protocol version 5
[Mon Jan 25 12:12:28 2016 804] Abnormal connection closed to ser:device=/dev/rfcomm1;baud=115200;stop=1;parity=none;fc=none;bits=8 : File write I/O error. Transport endpoint is not connected
[Mon Jan 25 12:12:28 2016 804] 0xa010 get node description from 0 : dest 30 protocol version 5
[Mon Jan 25 12:12:29 2016 719] 0xa011 list nodes from 0 : protocol version 5
[Mon Jan 25 12:12:29 2016 719] 0xa000 presence from 0 : protocol version 5
[Mon Jan 25 12:12:30 2016 719] 0xa011 list nodes from 0 : protocol version 5
[Mon Jan 25 12:12:30 2016 719] 0xa000 presence from 0 : protocol version 5
[Mon Jan 25 12:12:31 2016 718] 0xa011 list nodes from 0 : protocol version 5
[Mon Jan 25 12:12:31 2016 719] 0xa000 presence from 0 : protocol version 5

When a bluetooth connection is loss, it is still impossible to recover it on medulla. I need to restart medulla to reconnect the target. So I wait the next version of medulla or asebahttp2 :)

stephanemagnenat, 2016-01-25:
Ok. It seems your Bluetooth driver sucks... I close this bug as it is not an Aseba issue. We'll try to have the medulla features inside asebahttp2 as soon as possible. Note that you can bridge your program through asebahttp2 so that you can restart medulla without restarting the other side, as asebahttp2 will try to reconnect to medulla if this connection is lost.

closed 2016-01-25 by stephanemagnenat