aseba-community/aseba #459: last open directory in thymioupgrader

mbonani, 2015-11-30:
thymioupgrader do not remember last directory.
it works in my own compilation and not in new buildfarm. 

This is not important.

mbonani, 2016-01-22:
Ok has you ask this is present on the last buildfarm it alwas open "Documents", but not on my own compilation, it remember correctly the last directorier of open hex. file.

stephanemagnenat, 2016-01-22:
Actually this is not a bug, this feature was never implemented...

So this is a non urgent wish, delegating to 1.6

stephanemagnenat, 2016-04-21:
Is it still relevant now that the official firmware is automatically downloaded?

stephanemagnenat, 2016-12-19:
@mbonani is it still relevant now that the official firmware is automatically downloaded?

mbonani, 2016-12-20:
This feature seems to work again now...

closed 2016-12-20 by mbonani