aseba-community/aseba #471: State toggling with VPL not working

TiborUdvari, 2016-01-13:
I'm trying to do a timer delayed toggle from `State A` to `State B` and then from `State A` to `StateB `indefinitely.

Here is an [image](http://i.imgur.com/HZ1020V.png) of what I'm trying to do. Here is the [aesl file](https://www.dropbox.com/s/bbo8lbdv3csrck3/ThymioBlink.aesl?dl=0) as well. 
Thymio should switch indefinitely from red to blue.

I don't think it's possible to do this only using VPL. 
(which I'm trying to do because I'm doing a workshop for kid)

Do you think there would be a fix for this? Like an `else-if` statement in this block or the state copying and `return` statement in if block? 

I'm not sure if this would have any consequences in other use cases though. 

What do you think? 

```
onevent timer0
	timer.period[0] = 0
	if state[1] == 1 then
		call leds.top(32,0,0)
		new_state[1] = 1
		new_state[3] = 1
	end
	if state[1] == 1 and state[3] == 1 then
		call leds.top(0,0,32)
		new_state[1] = 1
	end

	call math.copy(state, new_state)
	callsub display_state
```

motib, 2016-01-13:
Hi Tibor,

1. A gray segment of a state means "ignore", but to distinguish the two states you need to make a segment white which means "off". What happened is that the event dependent on the state with only the upper right segment "on" (orange) will also fire when the state has two segments "on" (orange). You need to define the two states as (gray, orange, white, gray) and (gray, white, orange, gray) or something similar. The program in Figure 8.1 in the VPL tutorial demonstrates how this can be done.

2. You need to reset the timer with a timer _action_ block after a timer _event_ has occurred. Otherwise, the color will change only once.

I hope this helps,
Moti

TiborUdvari, 2016-01-13:
Thanks @motib 

I figured it out in the meantime. It's simple once I understood that gray meant "any". It was not very intuitive at first. Works just fine. I think calling it a state was what got me confused. Looking more into the code it makes sense as well.

closed 2016-01-13 by TiborUdvari