aseba-community/aseba #473: Crash when TAB with full lines

cbarraud, 2016-01-26:
Hi

My Aseba crashes when I try to indent several lines that touches the right border of the text-code area.

Here is an example:
[image]
Here I want to indent the couple of line that are highlighted. One hit on the TAB button of my keyborad is OK:
[image]
Two, three... until it reaches the right border, perfectly OK:
[image]
But here, when I hit TAB, Aseba freezes and crashes:
[image]

It is dependant on the size of the font. Here, I made the font size bigger (Ctrl + mouse wheel) so that the line of the if is already on two lines:
[image]
and I cannot hit TAB even once, the first hit kills Aseba.

I even tried a simpler code with just two ridiculously long named variables and it freezes instantly:
[image]

My Aseba is the 1.4.98 (build git-2868b3e/protocol ver.5) with Dashel 1.1.0. The Thymio connected to it was a Thymio II flashed with firmware V9.

Can anyone reproduce that? (my computer already did some strange things with the installer, but it was probably my anti-virus, maybe it's only my computer's problem ^^)

cbarraud, 2016-01-26:
I tested one more thing out of curiosity. With TABing only one line at a time, there is no problem.

stephanemagnenat, 2016-01-26:
I confirm this also leads to a freeze here on Linux, but not a crash.

stephanemagnenat, 2016-01-26:
It seems the problem is due to an infinite loop in `AeslEditor.cpp:778`

cbarraud, 2016-01-26:
On my computer, it freezes and then, if I try to do something else, it pops up the classical "Aseba stopped working" and then Windows tries to help (or not...)

stephanemagnenat, 2016-01-26:
Corrected in commit 9469c45.

closed 2016-01-26 by stephanemagnenat