aseba-community/aseba #474: open 1.4 VPL file in 1.5

mbonani, 2016-01-26:
There is a bug using a file that was generate by 1.4 with VPL code.
At first time the code is note regenerated and the "PaireExecuted" event is not present, so the visual feedback is not working. Then when you modify something the code is update and "PaireExecuted" is added but there is no global event added and VPL generate an wrong code.
Here is the file I use to test: 

[VPL-piste-code_1-4.zip](https://github.com/aseba-community/aseba/files/105665/VPL-piste-code_1-4.zip)

stephanemagnenat, 2016-01-27:
Does this problem happens with ThymioVPL, Aseba Studio, or both?

stephanemagnenat, 2016-01-27:
I found the source of the problem, so I assume it happens in Studio only. Commit e7e13a4 attempts at correcting it. As I have no Thymio with me (I sent it to Cambridge this morning) I cannot test.

mbonani, 2016-01-27:
Problem only in Studio
Your patch improved ;  "PaireExecuted is add in studio openning the file. The code is still unchanged before you click in the paired window. It is possible to click play with the code that is stored in the file.

stephanemagnenat, 2016-01-29:
Should be fixed by commit 4170160

closed 2016-01-29 by stephanemagnenat