aseba-community/aseba #475: Wireless Thymio Network Configurator cannot be open a second time

mbonani, 2016-01-27:
Netconfig, is working, but if you quit and want to relaunch it freeze. You have to unplug the dongle and restart. It seems it let the port in a strange state. 
You can launch after a netconfig a studio or VPL and its works, but relaunching netconfig afterwards it freeze.

stephanemagnenat, 2016-01-29:
That is strange. Do you confirm this problem appears only on OS X ?

mbonani, 2016-01-29:
In studio MainWindow.cpp line 876:
 `write16(file, ASEBA_PROTOCOL_VERSION);`
It should be the current node protocol version. 
I try to fix myself, but I have to take course of C++

mbonani, 2016-01-29:
only on mac:
lldb: (lldb) run
Process 65642 launched: '/Users/administrator/jenkins/workspace/aseba.osx/build/packager/dmg_contents/Aseba/Wireless Thymio Network Configurator.app/Contents/MacOS/thymiownetconfig' (x86_64)
Process 65642 exited with status = 0 (0x00000000) 
(lldb) run
Process 65649 launched: '/Users/administrator/jenkins/workspace/aseba.osx/build/packager/dmg_contents/Aseba/Wireless Thymio Network Configurator.app/Contents/MacOS/thymiownetconfig' (x86_64)
Process 65649 stopped
* thread #1: tid = 0x153c56, 0x00007fff8ca9145e libsystem_kernel.dylib`read + 10, queue = 'com.apple.main-thread', stop reason = signal SIGTERM
    frame #0: 0x00007fff8ca9145e libsystem_kernel.dylib`read + 10
libsystem_kernel.dylib`read:
->  0x7fff8ca9145e <+10>: jae    0x7fff8ca91468            ; <+20>
    0x7fff8ca91460 <+12>: movq   %rax, %rdi
    0x7fff8ca91463 <+15>: jmp    0x7fff8ca8ac02            ; cerror
    0x7fff8ca91468 <+20>: retq   
(lldb) ^C
(lldb)

mbonani, 2016-01-29:
truss
	PID/THRD  SYSCALL(args) 		 = return
Date/Time:       2016-01-29 11:13:46 +0100
OS Version:      Mac OS X 10.11.3 (Build 15D21)
Architecture:    x86_64h
Report Version:  22

Command:         Wireless Thymio Network Configurator
Path:            /Users/USER/*/Wireless Thymio Network Configurator.app/Contents/MacOS/thymiownetconfig
Version:         ??? (1.2)
Parent:          launchd [1]
PID:             65788

Event:           hang
Duration:        1.10s (process was unresponsive for 16 seconds before sampling)
Steps:           11 (100ms sampling interval)

Hardware model:  iMac14,1
Active cpus:     4

Fan speed:       1401 rpm

--------------------------------------------------
Timeline format: stacks are sorted chronologically
Use -i and -heavy to re-report with count sorting
--------------------------------------------------

Heaviest stack for the main thread of the target process:
  11  start + 52 (thymiownetconfig + 17028) [0x100004284]
  11  main + 1260 (thymiownetconfig + 23596) [0x100005c2c]
  11  Aseba::Thymio

stephanemagnenat, 2016-01-29:
hum, I guess we need to test with a build that has debug informations, and for lldb we need to get the full backtrace with command `thread backtrace`

stephanemagnenat, 2016-02-02:
This bug seems specific to OS X 10.10

stephanemagnenat, 2016-02-03:
We did not find an easy solution, it seems to be an OS X driver bug. We have documented this problem, and we delegate the work-around to a later time.

mbonani, 2016-11-23:
Tested with last 1.5.4 and OS X 10.11 and seems to not be there anymore. I will make another test to confirm.

stephanemagnenat, 2016-12-19:
@mbonani, can we close this bug?

closed 2016-12-20 by mbonani