aseba-community/aseba #476: .abo protocole mismatch

mbonani, 2016-01-27:
You cannot make a correct .abo for older version of firmware with protocole 4. 
It seems it put the aseba protocole version and not the node protocole version inside the abo. It result that you make .abo that is not leasible by the node afterward.

stephanemagnenat, 2016-01-29:
> You cannot make a correct .abo for older version of firmware with protocole 4.

Could you please clarify the use case when there is a problem? I am not sure which version of Studio and the target you use when this problem appears. Also what do you mean by "You cannot make a correct .abo"?

stephanemagnenat, 2016-01-29:
I understand what you mean.

stephanemagnenat, 2016-01-29:
Should be corrected by commit 5ff98e8, please test and close if so.

mbonani, 2016-01-29:
works

closed 2016-01-29 by mbonani