aseba-community/aseba #477: thymio upgrader

mbonani, 2016-01-29:
the new upgrader do not work if you update a wireless Thymio, unplug it and plug a new one for massive updating. 
You have to close and reopen. 
This is totally normal because the nodeID is not constant.
If we can improve it could be nice but it is not mandatory

stephanemagnenat, 2016-01-29:
That is true, because currently Thymio Upgrader does not keep a connection open, as flashing is done in a separate thread for safety. A solution would be to recheck the nodeId just before flashing, within the flashing thread. I can see on Monday whether doing so can be done quickly. Currently I do not have a wireless Thymio with me.

stephanemagnenat, 2016-02-01:
Should be fixed by commit 9fb9f4e, please test and close if it works correctly.

stephanemagnenat, 2016-02-03:
Seems to work, closing

closed 2016-02-03 by stephanemagnenat