aseba-community/aseba #479: Can not upgrade on Mac OS X

dblanchet, 2016-02-07:
Hi,

I'm currently trying to upgrade my Thymio-II from V8 to V9 on myt MacBookPro (El Capitan, Aseba 1.4).

I tried to launch "Thymio Firmware Upgrader.app", but I get this error message: 
 
[image]

Note that I can start program from VPL

stephanemagnenat, 2016-02-07:
El Capitan changed the naming of the serial port (changing "-" into "_"), Aseba 1.5, to be released in the next two weeks, works around this problem.

I will check tomorrow whether the command line flasher works on my computer.

dblanchet, 2016-02-08:
Thank you for quick feedback. I'll give a try to the development version ASAP.

dblanchet, 2016-02-08:
OK, I was able to update to V9 with dev version (Aseba-136.dmg) of the "Thymio Firmware Upgrader.app". I don't know if (and how) I can help you with command-line behaviour.

Thank you for your help.

mbonani, 2016-02-22:
The command line written by dblanchet is incomplete: "reset" is missing at the end. 
After testing on all platform I think there is no problem with the falsher in command line. 
I let you close

closed 2016-02-22 by stephanemagnenat