aseba-community/aseba #480: Change values of IR sensors instead of Something or Nothing

cbarraud, 2016-02-08:
In Blockly for Thymio, a teacher asked me if we could add a feature so that we could type the value of the IR sensors in the if test instead of "detecting proximity".

motib, 2016-02-09:
I have experienced significant variation in the response of the IR sensors so I regard this feature as essential.

stephanemagnenat, 2016-02-09:
Ok, we can think about it for 1.6

mbonani, 2016-11-23:
Having play more with blockly I think it is already possible to use value. But you have to use more blocks for that. See this example: 

[image]

stephanemagnenat, 2016-12-19:
@cbarraud and @motib, does @mbonani's answer solves your issue?

motib, 2016-12-20:
I don't use Blockly. I was just supporting the issue that Christhophe raised, so I can't comment on the adequacy of the solution.

cbarraud, 2016-12-21:
The solution of Michael is indeed working. Closing the issue.

closed 2016-12-21 by cbarraud