aseba-community/aseba #482: upgrader version 1.5 hang at 90%

mbonani, 2016-02-19:
It happens one time to me and one time to a user...
The other problem is that make the robot stock in the bootloader and this new version is not able to deal with that like the old one.

stephanemagnenat, 2016-02-19:
Hum! I suggest we investigate that Monday, until now we delay the announce of release 1.5

mbonani, 2016-02-19:
yes for investigation, but it is not reproductible...
probably a timing problem...

stephanemagnenat, 2016-02-22:
A work-around is to use the command line, `asebacmd` to recover the robot once the problem occurred. To flash a firmware, use this command (example given on Ubuntu):
```
asebacmd -t "ser:device=/dev/ttyACM0" wusb 1 Thymio2-V10.hex reset
```
You have to run it about 500 ms after having reset the robot physically.

stephanemagnenat, 2016-04-19:
Is it still present?

stephanemagnenat, 2016-12-19:
@mbonani, can you check whether this bug is still present in 1.5.5?

mbonani, 2016-12-20:
I never got it again... Like I said probably a timing problem coming from the charge of the OS... 
I will close.

closed 2016-12-20 by mbonani