aseba-community/aseba #483: Studio 1.5 crashes when getting events

FrancescoMondada, 2016-02-22:
When running the following code on a Wireless Thymio or a Thymio connected by USB, under windows or macos, ASEBA Studio does not display the events and crashes. All is working fine under 1.4.
----
var speed=50
motor.left.target = speed
motor.right.target = speed

onevent prox
	when prox.ground.delta[0]&lt;400 do
		speed = speed+20
		motor.left.target = speed
		motor.right.target = speed
	end
	emit displayspeed speed</node>
-----
Of course one should define an event called "displayspeed" with one argument.

mbonani, 2016-02-22:
I confirm the problem on windows and macosx
If argument is different of zero it crash.
It is working with VPL on studio, but not on studio alone. 
I suggest a problem with the feedback.

stephanemagnenat, 2016-02-22:
I confirm this occurs on Linux as well.

stephanemagnenat, 2016-02-22:
Fixed in commit 20d9ba0, please try latest generated builds (maybe wait some minutes or check on the status page (http://lsropc28.epfl.ch:8080/view/monitor/) if you have EPFL VPN access).

mbonani, 2016-02-22:
All is fine  for all version
every thing is onlie for user

closed 2016-02-22 by mbonani