aseba-community/aseba #489: AUR PKGBUID

THS-on, 2016-04-04:
Hello,
I made a PKGBUILD for Arch Linux.
```pkgname=aseba
pkgver=1.5.2
pkgrel=1
pkgdesc="A set of tools which allow beginners to program robots easily and efficiently."
arch=('x86_64')
url="https://github.com/aseba-community/aseba"
license=('LGPL3')
depends=('boost' 'qt4' 'qt5-tools' 'qwt5' 'libgudev' 'libxml2')
source=("https://launchpad.net/~stephane.magnenat/+archive/wily/+files/aseba_1.5.2_amd64.deb")
sha256sums=('bb36f3c0e46bb994199add2add062c401d13aa392eafe60989d475b5c214087c')
package() {
	msg2 "Extracting data.tar.gz"
	cd "${srcdir}"
	tar -xJC "${pkgdir}" -f data.tar.xz
	install -dm755 "${pkgdir}/usr/bin"
	msg2 "symlink /usr/lib/libqwt5.so to /usr/lib/libqwt-qt4.so.5"
	ln -s /usr/lib/libqwt5.so "${pkgdir}/usr/lib/libqwt-qt4.so.5"
	msg2 "Add $USER to dailout with: groupadd dailout and usermod -aG dailout user" 
}
```
Did I forgot something? Will Aseba support Arch Linux?

Thanks,
THS-on

stephanemagnenat, 2016-04-19:
Hello,

Thank you for your contribution! I have no knowledge on Arch, so I cannot tell you whether you forgot something or not.

Anyway, feel free to edit this page https://www.thymio.org/en:linuxinstall to point to your package, and if you want me to include the package source file within the source tree, please fill a pull request.

kind regards,

Stéph

THS-on, 2016-04-24:
Hello,
At this time I have some problems with the Thymio Wireless dongle because it's mounted only for root.(No rw flag for user) Is there a way to solve this? (Maybe udev rule)
When I have a fully working package, I'll fill a pull request.
kind regards,
THS-on

stephanemagnenat, 2016-04-25:
I do not know for Arch, but for Ubuntu, the problem is that accessing `/dev/ttyUSBX` requires to be in the `dialout` group. You can check the Ubuntu information at https://www.thymio.org/en:linuxinstall.

THS-on, 2016-10-04:
Hello,
I have now a working version from that builds from the git sources. On Arch you have to add yourself to the uucp group.
Link: [https://gist.github.com/THS-on/d0c770ee2961fded21a73100fafc29ce]( https://gist.github.com/THS-on/d0c770ee2961fded21a73100fafc29ce)
For now only the binaries are packaged + the .desktop files. I'll add the documentation and libraries later.

stephanemagnenat, 2016-10-05:
Great! Do not hesitate to add this information to the Wiki on the Linux section (https://www.thymio.org/en:linuxinstall.) once you'll like to make it available to the larger public. Also, you can edit the help section (https://www.thymio.org/en:thymiosupport) to say that on Arch it is the uucp group, rather than the dialout group as in Ubuntu. When you change a page, please set the untranslated tags as explained here (https://www.thymio.org/contribute), as these will allow contributors to propagate your changes to other languages.

If you'll like to add your script to the main repository, you are welcome. I do not know if there is a standard repository for Arch like `debian` or `rpm` for Debian and Fedora. If you wish to do so, please fill a pull request against master.

Thank you for your contributions!

THS-on, 2016-10-05:
I uploaded my package to the AUR, so it's now public for everyone. I changed all linuxinstall pages to show Arch Linux now listed.
Link: [https://aur.archlinux.org/packages/aseba-git/]( https://aur.archlinux.org/packages/aseba-git/ )
Arch Linux AUR uses simple build instructions and you build the package locally.

THS-on, 2016-11-20:
I finished my AUR package today and it's now feature complete.
I'll make an non git build when a new version of Aseba is released.

closed 2016-11-20 by THS-on

stephanemagnenat, 2016-11-21:
Great, thank you very much! We are about to release 1.5.4 from the 1.5.x branch, I have tagged it already: https://github.com/aseba-community/aseba/tree/1.5.4

If you could build packages for it and put them on the Wiki it would be great :-)

Thank you very much!

stephanemagnenat, 2016-12-05:
I just released version 1.5.5, if you are interested to update your Arch Linux packages, it would be great!