aseba-community/aseba #491: Reconnection problem, doesn't go back to Blockly

cbarraud, 2016-04-13:
Hello

A teacher sent us a mail, I tested and I think there is something wrong with the reconnection process of Blockly.

The thing is, when you code something with Blockly, then disconnect your robot and reconnect it, it doesn't go back to Blockly, but to Aseba Studio. Here is my example:

[image]

I did this code, then unplugged the robot, everything is ok:

[image]

and then, when I re-plug Thymio, it goes back to Aseba Studio

[image]

I'm on Windows 8.1 (the teacher was on 7). Aseba version:

[image]

stephanemagnenat, 2016-04-13:
I confirm the problem, it also happens on Linux, I will look into it as soon as possible.

stephanemagnenat, 2016-04-14:
It is a bug in the Blockly integration indeed, it should be fixed by commit e8bbefdc84d21ba14240c1235a1e8742ff1fd838. Please check the generated build, and close if the problem is solved for you.

stephanemagnenat, 2016-04-18:
@cbarraud please close the issue if the problem has been solved on your side, thank you :-)

cbarraud, 2016-04-20:
It's ok on my side, thank you :+1: 

Closing the issue ;)

closed 2016-04-20 by cbarraud

mbonani, 2016-04-20:
test on mac Os 10.6 and 10.11 should work in between.