aseba-community/aseba #492: Overlay text in VPL

cbarraud, 2016-04-15:
Hi

there is an overlay text in VPL when you let your mouse on the blocks that displays the ID of the block: 

[image]

These messages could be way more useful if they were a bit more descriptive. Could we brainstorm about these descriptions?

# Proposition

##  Events
### Buttons
Make Thymio react when you press a button. Click on the grey button to choose which button to use.
### Proximity sensors
Make Thymio react when it detects something around its. Click on the grey sensor(s) to choose which one(s) to use. White and red means something in front; black means nothing in front.
### Ground sensors
Make Thymio react when it detects something on the ground.  Click on the grey sensor(s) to choose which one(s) to use. White and red means white surface; black means black surface or void.

...

##  Actions
### Motors
Choose this block to change Thymio's motors speed. Drag the two sliders to choose the speed of the left and right wheel.
### Top colour
Choose this block to change Thymio's top colour. Drag the three sliders to mix colours.

...

(to be continued) I'm not sure about the length of the messages... What do you think?

stephanemagnenat, 2016-04-15:
I think the messages are too long, plus I am not sure we can use rich-text formatting in the tooltip. Moreover, this would demand new translation work. I suggest to only put the description of the block.

stephanemagnenat, 2016-04-18:
I've implemented my suggested solution in commit b76fa62, using same names as in https://www.thymio.org/en:thymiovpl. Please test in English, if happy, I'll translate into other languages.

mbonani, 2016-04-20:
It is much better, I am not sure "event block" and "action block" in all flyover are useful

cbarraud, 2016-04-20:
Much better indeed. After reflection, it's probably better to keep it short.

I think that keeping the action or event block text is good. Or at least action and event. I would keep it like this personally.

cbarraud, 2016-04-20:
So, we discussed internally and, as the "Event blocks" and "Action blocks" are written above the columns, we can erase them from the tooltip messages. Thus, we remove "action block" and "event block" from the messages :)

Let's start to translate.

stephanemagnenat, 2016-04-20:
Implemented in commit 4a28816.

stephanemagnenat, 2016-04-21:
Translations completed in commit 8d2f9f3303cce7958e3b97be61a8a78c67a172db.

closed 2016-04-21 by stephanemagnenat