aseba-community/aseba #493: Text more user-friendly when deconnection or when Thymio is not plugged

cbarraud, 2016-04-15:
The texts of the pop-up windows when Thymio is not plugged is not that user-friendly. For example, ThymioVPL says that when you try to start it without a robot:

[image]

The Thymio Firmware upgrader is much more user-friendly:

stephanemagnenat, 2016-04-15:
This is already implemented in commit 71a23929d5c911579ee88edbf286c96d433a7453.

mbonani, 2016-04-15:
For this implementation is that here possible to add an image? If not, can you just add also the dongle.

stephanemagnenat, 2016-04-18:
Image should be easy to add, at worst we subclass the dialog, so yes :-)

Dongle added to text in commit d701e8e.

mbonani, 2016-04-20:
Ok for me, it is good or the moment, we can think about an image later.

closed 2016-04-20 by mbonani