aseba-community/aseba #500: One more note in the music block

cbarraud, 2016-04-26:
Hello

I am thinking about an activity in which you could encode letter in morse code thanks to the music block. 
The idea is to associate a sensor for example to a melody which would be a letter coded in morse. You could then encode words by using the different sensors or buttons or even by using states in advanced mode. 

For example, HELLO is coded like that in Morse:

H:  o o o o
E: o
L: o _ o o
O: _ _ _

(with o a dot and _ a dash)

In VPL, you could encode it like that: 

[image]

But the problem is that it's very quick, not easy at all to recognise a letter. The solution is to let one empty note between each dash or dot and that works pretty well. 

[image]

The problem is that there are 6 notes, so with one blank between each, we arrive at 3 notes max. Morse code has letters coded with 4 bits,,, So if we added one note in the VPL music block, we could do easy Morse code :)

How would that impact on the rest...? Would that be difficult to implement? Just wondering ;)

motib, 2016-04-26:
Hi Christophe,

I actually learned Morse 50 years ago when it was used in amateur radio. I think I could receive 15-20 words/minute :-).

In the VPL tutorial, Exercise 6.1 asks for an implementation of morse code. In the answer directory I have a program, but it was written when there was no "rest" just long and short notes.

By definition a "dash" is three times as long as a "dot" (but I forgot how long the space between letters is, perhaps twice as long as a dot). Currently, the white is only twice as long as the black note, so I used multiple notes for each dot and dash.

I like your solution but I don't think it is practical since there aren't ~40 different sensors for the different letters, numbers, punctuation. In any case, if a change is made, I would make the white note three times as long as the black note because the current difference is hard to make out.

Moti

cbarraud, 2016-12-21:
In the end, we did it in another way for the Thymio Challenge Pack and, as it would not be that a difference, I think we can put energy on more important matters :)

Closing the issue

closed 2016-12-21 by cbarraud