aseba-community/aseba #527: fedora packages are missing executable

marvelous, 2016-06-07:
I have installed aseba-1.5.3-1.fc22.x86_64, but it doesn't contain the asebastudio executable. `rpm -ql` shows:

```
/usr/bin/asebacmd
/usr/bin/asebadummynode
/usr/bin/asebadump
/usr/bin/asebaexec
/usr/bin/asebahttp
/usr/bin/asebahttp2
/usr/bin/asebaplay
/usr/bin/asebarec
/usr/bin/asebaswitch
/usr/share/applications/asebachallenge.desktop
/usr/share/applications/asebaonlinehelp.desktop
/usr/share/applications/asebaplayground.desktop
/usr/share/applications/asebastudio.desktop
/usr/share/applications/asebastudiothymio.desktop
/usr/share/applications/thymioupgrader.desktop
/usr/share/applications/thymiovpl.desktop
/usr/share/applications/thymiownetconfig.desktop
/usr/share/doc/aseba
/usr/share/doc/aseba/README.Debian
/usr/share/doc/aseba/challenge-goto-energy.aesl
/usr/share/doc/aseba/changelog
/usr/share/doc/aseba/copyright
/usr/share/doc/aseba/readme.md
/usr/share/doc/aseba/unifr.playground
/usr/share/icons/hicolor/48x48/apps/asebachallenge.png
/usr/share/icons/hicolor/48x48/apps/asebahelp.png
/usr/share/icons/hicolor/48x48/apps/asebaplayground.png
/usr/share/icons/hicolor/48x48/apps/asebastudio.png
/usr/share/icons/hicolor/48x48/apps/asebastudiothymio.png
/usr/share/icons/hicolor/48x48/apps/thymioupgrader.png
/usr/share/icons/hicolor/48x48/apps/thymiovpl.png
/usr/share/icons/hicolor/48x48/apps/thymiowebbridge.png
/usr/share/icons/hicolor/48x48/apps/thymiownetconfig.png
/usr/share/icons/hicolor/scalable/apps/asebachallenge.svg
/usr/share/icons/hicolor/scalable/apps/asebahelp.svg
/usr/share/icons/hicolor/scalable/apps/asebaplayground.svg
/usr/share/icons/hicolor/scalable/apps/asebastudio.svg
/usr/share/icons/hicolor/scalable/apps/asebastudiothymio.svg
/usr/share/icons/hicolor/scalable/apps/thymioupgrader.svg
/usr/share/icons/hicolor/scalable/apps/thymiovpl.svg
/usr/share/icons/hicolor/scalable/apps/thymiowebbridge.svg
/usr/share/icons/hicolor/scalable/apps/thymiownetconfig.svg
```

brettle, 2016-06-13:
This should be fixed now.

stephanemagnenat, 2016-06-14:
Thank you very much!

closed 2016-06-14 by stephanemagnenat