aseba-community/aseba #531: uncussitent behavior when using stop button and sound card

mbonani, 2016-06-21:
A user make attention with simple code using sound. If you use in VPL this code:
[image]
First time you download the code thymio is quite and then go forward at first sound. 
Noise of motor make fire again and again the sound event. But when you click stop and play again Thymio go forward directly, without waiting a sound level. It seems that it keep a sound event in memory...
When you do it in text mode using this code generated:
`# setup threshold for detecting claps
mic.threshold = 250
# reset outputs
call sound.system(-1)
call leds.top(0,0,0)
call leds.bottom.left(0,0,0)
call leds.bottom.right(0,0,0)
call leds.circle(0,0,0,0,0,0,0,0)

onevent mic
	motor.left.target = 500
	motor.right.target = 500
	emit pair_run 0

onevent prox
	when prox.horizontal[2] >= 2000 do
		motor.left.target = 0
		motor.right.target = 0
		emit pair_run 1
	end
`
you have the same behavior if you press "stop thymio->load->run"
but not if you press "load->run" directly sometime it fire but not every time 

why when we "stop" the Thymio each time it fire the sound event? It is also perhaps a bug in Thymio firmware, but I wanted to be sure it is not in the VM.

stephanemagnenat, 2016-07-04:
This is most likely a bug in Thymio 2 firmware, see issue aseba-community/aseba-target-thymio2#16.

mbonani, 2016-09-20:
this issue is resolved in thymio firmware

closed 2016-09-20 by mbonani