aseba-community/aseba #533: An event "onstartup" is missing

basilstotz, 2016-06-26:
Hello,

there is no no event "onstartup", which is fired when the program is started. My wish is to add this event for educational reasons.

This new event would not add additional functionality, as this can be done just with 'free' code at the beginning of the program. But: Adding code outside events brakes the rule "everything is an event".

regards

Basil

stephanemagnenat, 2016-06-27:
I see your point, indeed currently initial code is put in an implicit
onstartup event. Would could add the possibility to specify onstartup
anywhere in the script and disallow code in the beginning, but we would
need to keep the current behavior for compatibility. Doing such
incremental additions that add two ways of doing things is not really
good for the users who might get confused. So my suggestion is to
collect desires for the Aseba script language, and do a larger change at
some point, maybe without backward compatibility or with explicit
versioning of the language.

basilstotz, 2016-06-28:
I think waiting for a bigger update this is good idea as it this is just a beautifier for the 'purity of the concept'