aseba-community/aseba #534: Textprogramming should have a mode where all tabs a synchronized

basilstotz, 2016-06-26:
Hello,

when programming for several identical nodes (i.e. all nodes have the identical code) it is not easy to manage that the code is really identical for all nodes. Every change in one tab has to be duplicated in all others.

I think it be nice to have a editor mode which keeps the content of all tabs identical.

regards 

Basil

stephanemagnenat, 2016-06-27:
I agree, the way I see this would be an option in the edit menu, called
"Propagate to similarly-named nodes" that copy the currently-edited code
to other nodes of same name. Do you like this approach?

basilstotz, 2016-06-28:
Thx, that's exactly what I thought.

stephanemagnenat, 2016-06-28:
I do not have the engineering resources to work on that right now, but pull requests are welcome :-)